GridProtectionAlliance/openPDC #2: openPDC Manager / Configure Security / Manage User Accounts - Cannot Save Changes to Existing Account

ajstadlin, 2016-04-10:
Steps to Repeat
1) Install openPDC Manager, SQLite Database, Database Authentication Account
2) Run openPDC Manager, Select the System / Configuration Security
3) In the Manage User Accounts section, select the Account created during the installation.
4) Change the expiration date so the account will not expire in a few months.
5) Click the Save icon or text link.
The result will be a "Save UserAccount Exception:" dialog with the following message:
Exception has been thrown by the target of an invocation.  
Inner Exception:  User name already exists.

Configuration Details:
openPDC Manager version 2.2.1.0 using SQLite
openPDC version 2.1.59.0 (built on target platform Debian 8.3)
Cross Platform:  openPDC Manager in Windows 10 and openPDC server running in Debian 8.3.
openPDC Manager accessing the openPDC server database via Samba File Share.

//AJ

StephenCWills, 2016-04-11:
Confirmed the issue in the latest nightly build (v2.2.16).

StephenCWills, 2016-04-11:
A resolution has been implemented in gsf and tested against the latest nightly build. This fix should work for any version of the openPDC since v2.1.177.

https://github.com/GridProtectionAlliance/gsf/commit/e7f69846726537000a9260a8c34d1196a78dff37

For version 2.1.59, you can modify the same line of code as follows:

`existing = Convert.ToInt32(database.Connection.ExecuteScalar(database.ParameterizedQueryString("SELECT COUNT(*) FROM UserAccount WHERE Name = {0} AND ID <> {1}", "name", "id"), DefaultTimeout, userAccountSID, database.Guid(userAccount.ID)));`

closed 2016-04-11 by StephenCWills

StephenCWills, 2016-04-11:
One more thing I may need to clarify. The issue and the resolution are both limited to the openPDC Manager. You do not need to apply the fix and rebuild the openPDC on your Debian 8.3 server. However, if you find that you need to use a matching version of the openPDC Manager, the fix I mentioned for version 2.1.59 can be applied to a custom build of the openPDC Manager.