GridProtectionAlliance/openPDC #3: POSIX install-openPDC.sh Script Fails for Nightly Builds

ajstadlin, 2016-04-16:
Steps to Repeat
1.  Download the POSIX openPDC installation script:  [install-openPDC.sh](http://www.gridprotectionalliance.org/Products/openPDC/Scripts/install-openPDC.sh)
2.  Run the script with the nightly builds parameter -n
`sudo bash install-openPDC.sh -n`
3.  The script fails starting at the point where it tries to run the subscript _buildmono.sh_
`Building Grid Solutions Framework for Mono...`
`install-openPDC.sh: line 248: cd: gsf/Source: No such file or directory`
`sh: 0: Can't open buildmono.sh`

The results details are in the attached text file.
[openPDC_POSIX_Installation_NightlyBuilds_Fails.txt](https://github.com/GridProtectionAlliance/openPDC/files/222031/openPDC_POSIX_Installation_NightlyBuilds_Fails.txt)

Note:  Running `sudo bash install-openPDC.sh -p` properly installs the stable release in my same platform.

Platform:  Debian 8.4.0 amd64 virtual machine running in Windows Server 2012 R2 HyperV
Debian 8.4.0 minimal install from netinstall iso.  Prerequisites installed per **Deploying openPDC on POSIX Platforms** Apr 8, 2015

StephenCWills, 2016-04-18:
Confirmed in Debian 9.

The issue can be traced back to before line 248.

```
Downloading Grid Solutions Framework source code...

--2016-04-15 22:18:13--  http://www.gridprotectionalliance.org/NightlyBuilds/GridSolutionsFramework/Beta/GridSolutionsFramework.Source.zip
Resolving www.gridprotectionalliance.org (www.gridprotectionalliance.org)... 68.169.166.227
Connecting to www.gridprotectionalliance.org (www.gridprotectionalliance.org)|68.169.166.227|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 71863404 (69M) [application/x-zip-compressed]
Saving to: ‘GridSolutionsFramework.Source.zip’

GridSolutionsFramework.S 100%[===================================>]  68.53M  4.36MB/s   in 17s

2016-04-15 22:18:29 (4.15 MB/s) - ‘GridSolutionsFramework.Source.zip’ saved [71863404/71863404]

Extracting Grid Solutions Framework source code...

unzip:  cannot find or open Source.zip, Source.zip.zip or Source.zip.ZIP.
rm: cannot remove ‘Source.zip’: No such file or directory
```

There seems to be a discrepancy in the name of the archive containing the GSF source code. I made a minor modification to install-openPDC.sh to fix that, however that does not seem to be the only problem with the nightly build so I won't be closing this issue just yet.

StephenCWills, 2016-04-18:
First problem is the Mono compiler does not support v4.6 as a target framework. Every *.csproj file needs to be modified to target v4.5 for the Mono build. We have yet to roll out an official fix for this, but here is the command I ran on my system to do the replacement.
```
find -name *.csproj -execdir sed -i 's/<TargetFrameworkVersion>v4.6<\/TargetFrameworkVersion>/<TargetFrameworkVersion>v4.5<\/TargetFrameworkVersion>/' '{}' \;
```

Second problem: System.Net.IPAddress.MapToIPv6() is missing from the Mono API. We already rolled out a fix for this, but the nightly build hasn't been updated yet. Here is the commit implementing the fix.
https://github.com/GridProtectionAlliance/gsf/commit/db63aaa9498e645206fc4d3e839d1a928c7d87ce

I am still unable to build GSF.PhasorProtocols. At a glance, this seems to be related to a refactoring that changed the casing on a few of the namespaces in GSF.PhasorProtocols a while back.
https://github.com/GridProtectionAlliance/gsf/commit/138970468e58138d9680f154d13b1a5eb3066b15

ajstadlin, 2016-04-20:
Thank you for the response to this issue.  I look forward to the next stable release.

ritchiecarroll, 2016-04-23:
OK - finally got this going. Had to exclude some newly added projects from the MONO configuration and change one project to not reference WPF based UI assemblies (which it should not have in the first place). Also, had to use "MONO_IOMAP=case ./buildmono.sh" prefix to get MONO to ignore file casing - I haven't tracked down the case-sensitive file differences in the csproj files vs what is checked into Git yet, but the MONO_IOMAP=case resolves. I will go ahead add this to the install-openPDC.sh bash script for any calls to xbuild to make sure this is not an issue in the future -- as soon as I have a decent Internet connection. You should be good to go now with latest check-ins and the MONO casing command above. Note that Stephen's current sed command to change csproj files from .NET 4.6 to .NET 4.5 is also still required. Guess I will add that to the install scripts as well for the time being, at least until MONO supports .NET 4.6 references.

StephenCWills, 2016-04-23:
I figured out how to configure my repo to pick up case changes in folder names.
`git --config core.ignorecase false`

As a result, I was able to fix GSF.PhasorProtocols.
https://github.com/GridProtectionAlliance/gsf/commit/ce3736a1c671a83a1224671d34266ad3d08f8180

StephenCWills, 2016-04-25:
I just made the necessary updates to the [install-openPDC.sh](http://www.gridprotectionalliance.org/Products/openPDC/Scripts/install-openPDC.sh) script to close the gap to getting this working. My test just now verified that the installation process has been fixed so I'm closing the issue. Please test and verify that all issues have been resolved on your end. Thanks!

closed 2016-04-25 by StephenCWills

ajstadlin, 2016-05-07:
Thank you for the updated install-openPDC.sh script.  Sorry for the delay getting back to you.
I verified that the script is updated with the above fixups for MONO_IOMAP=case and framework 4.5 version is set in the csproj files.
However, my installation from nightly builds still does not complete properly.  I admit I am somewhat of a Linux novice, but I wanted to give it a try.  I can still continue my Pi project using the stable 2.1.59 build.  I just wanted to let you know it still does not work out of the box with the nightly build.  FYI:  I'm up to page 13 in the documentation for this setup:  [openPDC_Debian84_Deployment_Instructions_2016](https://github.com/ajstadlin/openPDC/blob/aj-2016-05/Source/Documentation/wiki/Running_openPDC_on_Linux_and_Mac.files/openPDC_Debian84_Deployment_Instructions_2016.pdf)

Attached are my console log and the install-openPDC.log.

StephenCWills, 2016-05-08:
I just fixed the error I found in your log file. The nightly build should be working again tomorrow. For what it's worth, I also discovered that the build was probably working right up until the day before you tried it.

ajstadlin, 2016-05-11:
Thank you for the updated install-openPDC.sh script.  I just ran it (5PM EST 05/11/16) with the -n -p parameters and it worked!  I guess the script issue can stay closed B-)

However, there is an issue with the nightly build, probably not related to the install-openPDC.sh script.  I'll open a new issue for that.

Thanks for your hard work!!