GridProtectionAlliance/openPDC #4: error adding pmu in outputstream

pietropau, 2016-04-27:
When you setup new concentrated outputstream, an exception has been thrown during pmu adding  wizard .

Openpdc version:   2.2.29.0
.Net Runtime Version:  4.0.30319.42000
Mysql version: 5.7.12
Mysql connector NET version: 6.9.8
OS: Windows 7 professional

[image]

[error_concentrator_output_stream.txt](https://github.com/GridProtectionAlliance/openPDC/files/238395/error_concentrator_output_stream.txt)

Thanks,
Pietro

ritchiecarroll, 2016-04-27:
Looks like an issue with MySql, we will take a look...

StephenCWills, 2016-04-27:
This issue caused by the introduction of the VIRTUAL reserved keyword in MySQL version 5.7.6.
http://dev.mysql.com/doc/refman/5.7/en/keywords.html#ftn.idm140039936521520

This issue has been fixed in the latest code for GSF.
https://github.com/GridProtectionAlliance/gsf/commit/3dbac691bd2f03e565c591be07539a4b748647c5

The fix will show up in the next nightly build of openPDC (v2.2.30).

closed 2016-04-27 by StephenCWills

ritchiecarroll, 2016-04-27:
BTW - nightly builds occur every night (from a US Eastern time-zone perspective) any time there have been check-ins during the day. You can download the openPDC nightly build from here:

https://www.gridprotectionalliance.org/NightlyBuilds/openPDC/Beta-VS2012/Synchrophasor.Installs.zip