GridProtectionAlliance/openPDC #5: OPENPDC DATA MISSING

cosimopis, 2016-05-06:
Hallo,

we are currently testing OpendPdc in Terna, the Italian Transmission System Operator. 
One issue encoutered is related to some data missing systematically. I tried to modify the parameter configuration of the PMU into "Manage Device Configuration" mask (e.g. delayed connection interval and so on). Unfortunately no benefits have been achieved.
Could you provide support to fix?
Here in the following an example of data exporting:

Time,TERNA_BULM-IT:F Bulciago Frequency,TERNA_BULM-IT-PM1:V Bulciago Sb A 380kV  + Voltage Magnitude,TERNA_BULM-IT-PA1:VH Bulciago Sb A 380kV  + Voltage Phase Angle,TERNA_CHGP-IT:F Chiaramonte Gulfi Frequency,TERNA_CHGP-IT-PM1:V Chiaramonte Gulfi SbaA380  + Voltage Magnitude,TERNA_CHGP-IT-PA1:VH Chiaramonte Gulfi SbaA380  + Voltage Phase Angle
03-mag-2016 03:54:20.600,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.620,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.640,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.660,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.679,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.700,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.720,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.740,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.760,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.780,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.800,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.820,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.840,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.860,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.880,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.900,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.920,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.940,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.960,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:20.980,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.000,49,999,402888,3,138,1064,49,997,228747,6,-158,9617
03-mag-2016 03:54:21.020,50,005,402794,8,138,2601,50,003,228893,4,-158,8526
03-mag-2016 03:54:21.040,50,005,403055,3,138,3344,50,006,228841,3,-158,8139
03-mag-2016 03:54:21.059,50,003,402956,3,138,4044,50,001,229064,9,-158,7852
03-mag-2016 03:54:21.080,50,002,403205,8,138,4497,50,003,228869,2,-158,7381
03-mag-2016 03:54:21.100,50,002,402852,6,138,5219,50,001,229042,3,-158,6989
03-mag-2016 03:54:21.120,50,001,402844,5,138,562,50,001,228896,7,-158,6342
03-mag-2016 03:54:21.140,49,997,403107,8,138,5197,50,002,228875,7,-158,5761
03-mag-2016 03:54:21.160,50,005,403067,8,138,6038,50,005,228819,6,-158,4975
03-mag-2016 03:54:21.179,49,999,403324,5,138,5469,50,002,229000,4,-158,4255
03-mag-2016 03:54:21.200,50,402922,6,138,565,50,006,229006,8,-158,3082
03-mag-2016 03:54:21.220,50,002,403063,138,5346,50,002,228983,2,-158,2384
03-mag-2016 03:54:21.240,50,402984,9,138,5317,50,004,228843,6,-158,2132
03-mag-2016 03:54:21.260,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.280,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.300,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.320,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.340,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.360,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.380,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.400,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.420,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.440,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.460,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.480,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.500,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.520,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.540,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.559,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.580,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.600,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.620,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.640,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.660,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.679,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.700,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.720,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.740,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
03-mag-2016 03:54:21.760,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale,Non un numero reale
[... cut ...]
03-mag-2016 03:54:23.160,49,997,402738,1,137,0699,49,992,228543,2,-160,8369
03-mag-2016 03:54:23.179,49,997,402511,9,136,9837,49,997,228749,4,-160,9434
03-mag-2016 03:54:23.200,50,402702,4,136,9245,50,229136,6,-160,9989
03-mag-2016 03:54:23.220,49,992,402824,3,136,7857,49,997,229079,5,-161,1209

Many thanks,

Cosimo Pisani 
Analisi di esercizio 
Gestione Ingegneria Sistema Elettrico
Dispacciamento e Conduzione
Terna Rete Italia SpA - Via Palmiano, 101 - 00138 Roma (RM) - Italia 
Tel. 06-83155773  Cell. +39 3277085577
www.terna.it

StephenCWills, 2016-05-06:
Hello Cosimo,

Can I assume the data export came from the Historian Trending Tool? What happens if you uncheck the "Align timestamps" checkbox in the export dialog?

[image]

Thanks,
Stephen

pietropau, 2016-05-09:
Hi Stephen,
The data export came from historian tool with "Align timestamps" checked (50 frames per second).
When "Align timestamps" is uncheck  doesn't exist missing data but the timestamp are strange.
attached export file with both options.

[export_uncheck.csv.pdf](https://github.com/GridProtectionAlliance/openPDC/files/254950/export_uncheck.csv.pdf)
[export_check.csv.pdf](https://github.com/GridProtectionAlliance/openPDC/files/254949/export_check.csv.pdf)

Ciao
Pietro

StephenCWills, 2016-05-09:
Hi Pietro,

While the timestamps are consistent in each second, the interval between timestamps is not consistent between samples. This indicates that the interval between sample counts reported in the device's data frames is inconsistent, which suggests that this is not an issue with the parser in the openPDC.

I am assuming, based on some of our prior discussions, that these devices are ABB RES521 PMUs. Based on our experience with those devices at TVA, it is not uncommon for a device to begin reporting bad timestamps after it has been running for long periods of time. The only way we know of to fix it is to reboot the device. This may be the problem you are experiencing.

If it is indeed a problem with the device, I'm not sure if there is anything more we can do to help. However, if you can send us a configuration frame and capture file from the PMU Connection Tester when streaming data from one of your devices, we can take a look to see if we can provide any additional insight.

Thanks,
Stephen

cosimopis, 2016-05-10:
Dear Stephen,

thank you again for the support. I confirm you that we are considering PMUs ABB RES521 and, as you correctly stated, these devices performs very well (we never found any kind of problems). Therefore I suppose that it deals with something inside OpenPDC. For this reason I attach you, as required, both PMU configuration and capture file from PMU Connection Tester hoping you could discover something new.

Furthermore you find also two example of extraction with and without the use of “Align timetamps”.
Now, as you can effectively appreciated, the problem seems solved by disabling the option above and data are archived more or less regularly every 5 ms. On the contrary, by enabling “Align timetamps timestamps are regularly archived every 20 ms but with some regular loss of data (NaN).
We need some clarification about this last issue in order to be aware of what OpenPDC is doing. Could you provide some details at this regard?

Best regards

Cosimo

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

cosimopis, 2016-05-10:
Dear Stephen,

thank you again for the support. I confirm you that we are considering PMUs ABB RES521 and, as you correctly stated, these devices performs very well (we never found any kind of problems). Therefore I suppose that it deals with something inside OpenPDC. For this reason I attach you, as required, both PMU configuration and capture file from PMU Connection Tester hoping you could discover something new.

Furthermore you find also two example of extraction with and without the use of “Align timetamps”.
Now, as you can effectively appreciated, the problem seems solved by disabling the option above and data are archived more or less regularly every 5 ms. On the contrary, by enabling “Align timetamps timestamps are regularly archived every 20 ms but with some regular loss of data (NaN). 
We need some clarification about this last issue in order to be aware of what OpenPDC is doing. Could you provide some details at this regard?

Best regards

Cosimo

StephenCWills, 2016-05-10:
Cosimo,

The "Align timestamps" checkbox does a little bit of extra math in an attempt to line up timestamps with the sample rate of the device. In general, the calculation is designed to round timestamps down to the nearest frame timestamp. It also avoids displaying duplicate timestamps. Because all your timestamps appear to be smashed into the first 240 milliseconds at the top of each second, it's doing its best to assign those measurements to the 50 Hz timestamps at the top of the second. Because it suppresses duplicates, much of your data is being ignored by the exporter.

By unchecking "Align timestamps", you are essentially asking the Historian Trending Tools to display the actual timestamps as they were written to the historian. This makes no attempt to line up the timestamps so you may end up seeing duplicate timestamps or misaligned data, which can make the export difficult to analyze.

FYI, as a result of investigating this issue, we've found a slight problem with the logic that handles measurement timestamps in the openHistorian 1.0. Occasionally, due to arithmetic rounding errors with floating-point numbers, some precision is lost and the timestamps can be found to be incorrect, but the error is never more than 1 millisecond. However, it is causing some problems with the data export when aligning timestamps, so the current plan is to change the behavior of the data exporter to round to the nearest frame timestamp at the selected frequency. At the moment, this will have very little effect on your data exports because your timestamps do not appear to span the entire second.

Regarding the data export, the timestamps are indeed spaced approximately 5 ms apart, but occasionally are 6, 4, or even 3 milliseconds apart. In the IEEE 1344 parser, timestamps are calculated based on an integer sample count and the frame rate of the device. The only variable between data frames is the sample count. I don't know of any consistent interval by which an IEEE 1344 device can report a sample count that results in those kinds of variations. At worst under normal conditions, the difference in timestamps between frames should fluctuate between two values that are at most 1 millisecond apart. At 50 Hz, there shouldn't be any fluctuation at all. However, because the data in the historian has already been parsed by our parser, I can't dismiss the possibility that we are parsing the timestamps incorrectly, which is part of the reason I asked for a data capture from the PMU Connection Tester.

Regarding the data capture, it seems you responded to this thread via email and we didn't receive the attachments. If you are concerned about the sensitivity of the data, you can send the data via email to support@gridprotectionalliance.org and we will receive it.

Thanks,
Stephen

cosimopis, 2016-05-10:
Dear Stephen,

I sent the files not attached in the previous email. 
Some comments to your diagnosis of problem from our internal thoughts: 
we checked with PMU Connection Tester the received frames and noted that typically the time stamps of milliseconds is not rounded from PMUs. We wellknown from several years this fact and this was the idea in our existing “homemade” Wams concentrator: the sampling at receiving level (PDC) is every 20 ms. So in any case we round the samples at nearest integer (i.e. 00 ,20, 40, 60, 80 ms); so the questions are:

1.	Is it possible in OPENHIS round data in this way ? (if sampling of our system is 20 ms, this is, in our opinion)
2.	If we select “align timetamps” we expect that the math mechanisms that you described work, so no lack of data. In case of consequent samples duplication due to the rounding a possibility could be to shift to the nearest integer. i.e. if we receive 10:15:10.121 and 10:15:10.125, the first could be rounded to 10:15:10.00 and second to 10:15:10.20 . We don’t expect to receive more that two samples in 20 ms. In any case, it seems that “align timetamps” process doesn’t work in proper way because we loss every seconds 2/3 of data … and this doesn’t happen with existing Terna data concentrator
3.	Why the trend assumes always checked the option “align time stamps” and forbid the possibility to see data “as it is” ?
4.	Is it possible to extract the data as it acquired by PMU?

Finally here enclosed another PMU capture

Many thanks,

Cosimo

StephenCWills, 2016-05-10:
Cosimo,

Thanks for the PMU Capture. We were able to determine that this issue was caused by a bug in our IEEE 1344 parser after all. The sample capture showed a consistent interval of 20 in the sample count between data frames. The inconsistency in the timestamp interval of the data export was caused by arithmetic rounding errors encountered when writing to and reading from the historian.

1. The openHistorian does not round timestamps. It writes timestamps as-is, so the time that the device reported should be exactly what is stored by the historian.
2. The timestamp alignment shows 2/3 data loss because the timestamps which are stored in the historian are incorrect. The fix to the IEEE 1344 parser should solve that issue going forward.
3. The Historian Trending Tool merely checks the "Align timestamps" checkbox by default because we often see devices with skewed timestamps or missing data in practice. If you uncheck the "Align timestamps" checkbox, it will export the timestamps "as-is".
4. See above. If you uncheck the "Align timestamps" checkbox, it will export the timestamps "as-is".

The latest nightly build version (v2.2.42) now has fixes for all the issues described in this thread. Because we believe the issue to be fixed, I am closing the issue. Feel free to continue commenting and/or reopen the issue as necessary.

Thanks,
Stephen

closed 2016-05-10 by StephenCWills

cosimopis, 2016-05-11:
Hallo Stephen,

thanks for the prompt reply. In the new nightly build version it seems that the export process works correctly!!! 
Nonetheless we noticed some strange behavour of the waveforms in "Historian Data Viewer" by using "Trend" as if the improvements applied to the export process had no effect on the display.

Best regards,

Cosimo

StephenCWills, 2016-05-11:
Hello Cosimo,

When the timestamps were being parsed incorrectly, you should have seen a concentration of measurements at the top of each second with large gaps in between where there was no data. With the timestamps fixed, the data should be more evenly distributed so those large gaps should disappear. Therefore, the improvements should have had some effect on the display.

What you may be noticing is that the chart in the Historian Trending Tool is designed to downsample the data to speed up the visualization. Furthermore, the default chart resolution is set to a very low number: 100 samples. This means the tool will display a maximum of 100 samples on the chart by default. At 50 Hz, this can accurately display no more than 2 seconds of data. Increasing this number will improve the accuracy over larger time windows, but it may take a bit longer to update the chart.

If you still believe there is something wrong with the chart, it would help if I could see a screenshot.

Thanks,
Stephen

cosimopis, 2016-05-11:
Dear Stephen,

thanks for the kind answer…Now we understood the reasons of the “smooth” visualization.
Do you think this visualization could be customized to increase the number of points in the chart (for instance)? Or by fixing a sliding window with good visualization accuracy moving forward and backward on the basis of the user needs?

Thanks again,

Cosimo

ritchiecarroll, 2016-05-11:
On the front screen of the trending tool you can adjust the total number of points to display on the chart.

Ritchie