GridProtectionAlliance/openPDC #6: PowerCalculations.PowerStability

cosimopis, 2016-05-10:
Hallo,

I would like to have much more details about this algorithm since I wasn't  able to find support material.

Thanks a lot,

Cosimo

StephenCWills, 2016-05-10:
I'm not sure how much this will help, but here's a high-level description taken from the code comments.

> This algorithm calculates power and its standard deviation in real-time that can be used to determine if there is an oscillatory signature in the power output.
> If multiple voltage phasors are provided as inputs to this algorithm, then they are assumed to be redundant values on the same bus, the first energized value will be the voltage phasor that is used in the calculation.
> If multiple current phasors are provided as inputs to this algorithm, then they are assumed to be cumulative inputs representing the desired power output summation of the generation source.
> Individual phase angle and magnitude phasor elements are expected to be defined consecutively. That is the definition order of angles and magnitudes must match so that the angle / magnitude pair can be matched up appropriately. For example: angle1;mag1; angle2;mag2; angle3;mag3.

Ritchie may be able to provide more detail so I have assigned this issue to him.

ritchiecarroll, 2016-05-12:
This algorithm calculates a standard deviation of megawatts over a rolling time window. Any significant deviation will be a good "first-alert" indicator that an oscillation may be occurring. Many utilities will take the result of this calculation and send the value into SCADA and alarm when the value exceeds a threshold. Note that the ICCPExport built into the openPDC will allow you to create a file based link into your ALSTOM SCADA system if you wanted to use this tool in this manner.

closed 2016-05-12 by ritchiecarroll