GridProtectionAlliance/openPDC #8: Nightly Build, Installed on Debian 8.4 - Run As Console Mode is not Displayed

ajstadlin, 2016-05-11:
Run As Console mode is not displayed as expected after installing the nightly build (May 11, 2016 AM)

Steps to Repeat:
`sudo bash install-openPDC.sh -n -p`
`sudo mono /opt/openPDC/openPDC.exe -RunAsConsole`

After the -RunAsConsole command, it appears like Nothing happens.  However, openPDC is running as can be confirmed using a remote openPDC console and PmuConnectionTester on another machine.  Console commands can be typed OK.   This is similar to the behavior if quitting a working openPDC console in the same platform using Ctrl+C.

A summary of the platform setup procedure is described in the attached document.  The above issue occurs at step 14 in the procedure.

[openPDC_Debian84_Deployment_Instructions_2016--Appendix-Implementation_Summary.pdf](https://github.com/GridProtectionAlliance/openPDC/files/260248/openPDC_Debian84_Deployment_Instructions_2016--Appendix-Implementation_Summary.pdf)

StephenCWills, 2016-05-12:
We saw this exact same behavior in another of our applications after we introduced message filtering in the openPDC Console. Because openPDC Shell was interacting directly with the service host, it was attaching to events and sending messages without generating a client ID, which was breaking message filtering and causing some other strange behavior.

I checked in a fix just now, and confirmed it to fix the issue on Windows.
https://github.com/GridProtectionAlliance/gsf/commit/51d64af3a278589a898c4a7369eab020e23f82fd

I kicked off a build for v2.2.44 to test on Linux so that should be ready in 10 or 15 minutes. I'm closing the issue in the meantime.

closed 2016-05-12 by StephenCWills

ajstadlin, 2016-05-13:
Just writing to confirm this is fixed and works now.  I just tested with openPDC nightly build 2.2.46.
Running the install-openPDC.sh -p -n works now properly shows the console in SSH.
```
version

openPDC Service Version:

      App Domain: openPDCShell.exe, running on .NET 4.0.30319.17020
    Machine Name: openpdc-1
      OS Version: Unix 3.16.0.4
    Product Name: Debian GNU/Linux 8 (jessie) using Mono
  Working Memory: 108.496 MiB
  Execution Mode: 64-bit
      Processors: 3
       Code Base: opt/openPDC/openPDCShell.exe
      Build Date: 5/13/2016 4:03:30 PM
         Version: 2.2.46.0
```