GridProtectionAlliance/openPDC #11: Creating virtual output devices

Vtern, 2016-05-12:
Hallo,

I noticed that in creating virtual output devices it is not possible to include frequency measurements. Is it correct? How they can be provided (as analog for instance?).

Thanks again,

Cosimo

StephenCWills, 2016-05-12:
Hello Cosimo,

You are correct. The synchrophasor protocols assume that every device provides exactly one frequency in its output. As you suspect, it is possible to provide additional frequency measurements as analog signals. All you need to do is create new analog values for your virtual output device and then change the signal reference of the output measurements.

For example, if I want to map the frequency of a device named DEV2 to my virtual output device named VIRT, I would add an analog value to VIRT with load order 1. Then I would add DEV2-FQ to my output stream and change its signal reference to VIRT-AV1.

Thanks,
Stephen

closed 2016-05-12 by StephenCWills

Vtern, 2016-05-13:
Many thanks Stephen,

Regards,

Cosimo