GridProtectionAlliance/openPDC #9: ADAPTER CALCULATIONS RECURRENT FAIL

cosimopis, 2016-05-12:
Hallo,

I tested extensively some adapters both already available as default in OpenPdc and customized by myself. I found a recurrent error log which is a bit strange (I attach screenshoot of the log page and details of the newest) due to the type of calculations made.
Actually both (for instance) for "Manage power calculations" and a simply "DividebySqrt(3)" adapter it says: GSF.PhasorProtocols  --> System.DivideByzeroException

Both in power calculation and voltage conversion (phase to phase -->phase to ground) there are no division by zero?

Could you help us to fix?

Best regards,

Cosimo
[log_fail_detail.docx](https://github.com/GridProtectionAlliance/openPDC/files/260822/log_fail_detail.docx)
[image]

StephenCWills, 2016-05-13:
Hello Cosimo,

Your log file suggests this was a minor logic error in the IEEE 1344 frame parser that was related to the recent timestamp fix and caused by an incorrectly handled edge case. I've applied a fix for the edge case which I believe should fix the problem.

https://github.com/GridProtectionAlliance/gsf/commit/1c59ff7da59de78080b01a40047bd47a169fdd5d

The fix will be available in the next nightly build version (v2.2.46). I'm closing the issue for now. If you continue to encounter the error in the future, please reopen the issue.

Thanks,
Stephen

closed 2016-05-13 by StephenCWills

cosimopis, 2016-05-13:
Great!

Vtern, 2016-05-13:
When it will be available 2.2.46 version Stephen?

StephenCWills, 2016-05-13:
I'll kick off a build. It should be available in 10 or 15 minutes.

cosimopis, 2016-05-13:
Nice!
Thanks
Cosimo

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.