GridProtectionAlliance/openPDC #20: openPDCConsole in Windows Command Line not recognizing IPv4

ajstadlin, 2016-07-16:
**openPDCConsole Version 2.2.70** does not recognize IPv4 in the Windows Command Line.  For example:

`C:\GPA.Apps\openPDC>openPDCConsole.exe -server="openpdc-pi3:8500; interface=0.0.0.0"`
or
`C:\GPA.Apps\openPDC>openPDCConsole.exe -server="openpdc-pi3:8500" -interface="0.0.0.0"`

Fail with the result:

`Exception during connection attempt: IP address mismatch: unable to find an IPv6 address for "openpdc-pi3"`

However, it works OK if I set the connection string in the **openPDCConsole.exe.config** file:

`<add name="ConnectionString" value="Server=openpdc-pi3:8500; interface=0.0.0.0"
        description="Data required by the client to connect to the server."
        encrypted="false" />`

ritchiecarroll, 2016-07-16:
What happens if you don't put a space between 8500; and interface?

ajstadlin, 2016-07-16:
NO SPACE after the semi-colon Works!  I'll make sure the documentation update does not have a space!   

`openPDCConsole.exe -server="openpdc-pi3:8500;interface=0.0.0.0"`

THANKS!
//AJ

closed 2016-07-16 by ajstadlin