GridProtectionAlliance/openPDC #21: openPDCConsole doesn't work properly after upgrade to version 2.2

mlbeauchemin, 2016-07-19:
I have upgraded the openPDC software from v1.5 to v2.2 on Windows Server 2008 SP2. Upon starting the console, if I enter the 'list' command, it will work the first time, but if I try to enter another command, it will just hang and continue printing the 'State of process...' messages until I close and reopen the console. Stopping and restarting the service seems to get it to work normally for maybe a couple of hours, but then it stops working again.

The other issue is, when I open the console, it will fill up with 'State of process "HealthMonitor" has changed to "Processing"/"Processed".' messages and nothing else if left alone.

StephenCWills, 2016-07-19:
What happens if you enter the 'filter' command into the console?

mlbeauchemin, 2016-07-19:
I restarted the service 2 hours ago, so the console is still running normally, but when I enter 'filter', it says 'No filters defined.'

ritchiecarroll, 2016-07-19:
Couple of things could be going on, however, sounds like the system is having trouble re-authenticating you. Could be having difficultly getting to database, having trouble accessing domain controller (assuming you are authenticating via AD) or your AD account password may need to be changed., etc.  Would be helpful to view error log / status log after the event happens to further diagnose issue...

StephenCWills, 2016-07-21:
To clarify, if you can provide the following files, it would help us to diagnose the issue. All three files are in the directory where you installed the openPDC.

* ErrorLog.txt
* StatusLog.txt
* Status.txt

mlbeauchemin, 2016-07-21:
I'm trying to pinpoint exactly when the console stops working. I'd prefer to wait until it happens again before engaging your team in reviewing logs. I'll keep my eye on it until then and let you know. Thanks.

mlbeauchemin, 2016-08-01:
Since the initial problems which prompted this thread, I am unable to get the console to reproduce the issue. The console is still working properly after nearly 14 days since the last service restart. I suppose we can close the thread because I can't replicate the problem.

StephenCWills, 2016-08-01:
Let us know if you encounter the problem again, and we can reopen the issue.

mlbeauchemin, 2016-08-01:
Thanks, will do.

mlbeauchemin, 2016-08-25:
The console is having the issue again, though I don't know for sure when it started. I have provided ErrorLog.txt and StatusLog.txt going back to August 11. The Status.txt file hasn't been updated since 2014 and I didn't think that would be useful for you.
[ErrorLog.txt](https://github.com/GridProtectionAlliance/openPDC/files/437010/ErrorLog.txt)
[StatusLog.txt](https://github.com/GridProtectionAlliance/openPDC/files/437012/StatusLog.txt)

StephenCWills, 2016-08-25:
The StatusLog.txt file seems to have exactly one message that was written on the 11th of August. However, the ErrorLog.txt file has errors all the way up to 12:14 PM today. That seems highly unusual, so I'd just like to verify whether you can confirm that the StatusLog.txt file wasn't somehow clipped when you uploaded it or something along those lines.

mlbeauchemin, 2016-08-25:
That's the entire StatusLog.txt file.

StephenCWills, 2016-08-25:
Here is a list of questions for you.

* How many CPU cores does your system have?
* Have you restarted the service yet since you noticed this happening?
* If not, can you check the CPU usage on the system and let me know how it's doing?
* When this does happen, as far as you know, is the openPDC still operating normally? For instance, is it still able to archive data or export data to other systems?

mlbeauchemin, 2016-08-25:
I have not restarted the service yet. 
The CPU has 4 cores, and usage bounces around between 7% and 24%. 
We don't have this particular setup exporting the data to another system, but I believe it is still working normally; there are ppa_archive files in the Archive directory with current timestamps, at least.

StephenCWills, 2016-08-25:
Can you use the Historian Trending Tool to do a trend of openPDC memory usage over the period between August 11 and today?

1. Launch the Historian Trending Tool (HistorianView.exe in the openPDC installation directory).
2. Go to `File > Open archives...` and make sure that `...\Statistics\stat_archive.d` is one of the archives being opened. The Statistics folder should be located in the openPDC installation directory.
3. Type `DEFAULT!SYSTEM-ST3` into the search box at the top and click the `Select All` button next to the search box.
4. Make sure that the checkbox labeled `Choose an interval relative to current time` is unchecked.
5. Enter `8/11/2016 00:00:00.000` as the `UTC Start Time` and `8/25/2016 00:00:00.000` as the `UTC End Time`.
6. Enter `1000` into the text box labeled `Chart resolution`.
7. Go to `Options > Trend` to display the chart.

Here is a screenshot with the values entered into the correct fields.
[image]

I think it would help me to be able to look at that, so if you could take a snapshot of that chart and paste it into your response, I'd appreciate it.

Thanks,
Stephen

mlbeauchemin, 2016-08-25:
Looks like I can't go back that far. I can't even see 24 hours ago, and I don't know if there is a way to get it back.

[image]

StephenCWills, 2016-08-25:
What if you try to look at the week before the 11th?

mlbeauchemin, 2016-08-25:
There's nothing.
[image]

StephenCWills, 2016-09-15:
I'm still not sure what's causing this issue. The only thing I can think to do next would be for us to set up a test environment to replicate the issue for debugging. The problem with that is, we don't support Windows Server 2008 SP2, so we don't have matching environments available to us, and I've never seen this behavior on any other system. My best recommendation at this point would be to move your openPDC to a system with a newer operating system to see if that helps.

closed 2016-09-15 by StephenCWills