GridProtectionAlliance/openPDC #24: ACCESS TO DATA STORED BY OPENHISTORIAN

cosimopis, 2016-08-23:
Hi everyone,

The main issue is that right now we are not able to extract correctly data from the OpenHist Database. By using Historian trending tool we found these two situations referred to the case of use of “Align time stamp” option and not use of “Align time stamp. 
By using “Align time stamp” option  we have (50 sample/s is the chosen frame):
Historian Data Viewer Export: *-15m to *

Average,49,9954600816362
Maximum,50,1873931884766
Minimum,49,8472709655762

Time,TER_FIUMESANTO!IS:ST16 InputStream statistic for Latest actual mean frame rate for data received from input stream during last reporting interval.
08-ago-2016 12:56:22.720,Non un numero
08-ago-2016 12:56:22.740,Non un numero
08-ago-2016 12:56:22.760,Non un numero
08-ago-2016 12:56:22.780,Non un numero
08-ago-2016 12:56:22.800,Non un numero
08-ago-2016 12:56:22.820,Non un numero
08-ago-2016 12:56:22.840,Non un numero
08-ago-2016 12:56:22.860,Non un numero
08-ago-2016 12:56:22.880,Non un numero
08-ago-2016 12:56:22.900,Non un numero
08-ago-2016 12:56:22.920,Non un numero
08-ago-2016 12:56:22.940,Non un numero
08-ago-2016 12:56:22.960,Non un numero
08-ago-2016 12:56:22.980,Non un numero
08-ago-2016 12:56:23.000,Non un numero
08-ago-2016 12:56:23.020,Non un numero
08-ago-2016 12:56:23.040,Non un numero
08-ago-2016 12:56:23.060,Non un numero
08-ago-2016 12:56:23.080,Non un numero
08-ago-2016 12:56:23.100,Non un numero
08-ago-2016 12:56:23.120,Non un numero
08-ago-2016 12:56:23.140,Non un numero
08-ago-2016 12:56:23.160,Non un numero
08-ago-2016 12:56:23.180,Non un numero
08-ago-2016 12:56:23.200,Non un numero
08-ago-2016 12:56:23.220,Non un numero
08-ago-2016 12:56:23.240,Non un numero
08-ago-2016 12:56:23.260,Non un numero
08-ago-2016 12:56:23.280,Non un numero
08-ago-2016 12:56:23.300,Non un numero
08-ago-2016 12:56:23.320,Non un numero
08-ago-2016 12:56:23.340,Non un numero
08-ago-2016 12:56:23.360,Non un numero
08-ago-2016 12:56:23.380,Non un numero
08-ago-2016 12:56:23.400,Non un numero  

Without using “Align time stamp” option  we have:

Historian Data Viewer Export: *-15m to *

Average,50,0055549278688
Maximum,50,8752098083496
Minimum,49,8380470275879

Time,TER_BULCIAGO!IS:ST16 InputStream statistic for Latest actual mean frame rate for data received from input stream during last reporting interval.
08-ago-2016 13:02:50.487,49,84792
08-ago-2016 13:03:00.488,49,9974
08-ago-2016 13:03:10.489,49,98746
08-ago-2016 13:03:20.502,49,98782
08-ago-2016 13:03:30.502,49,98753
08-ago-2016 13:03:40.504,49,98742
08-ago-2016 13:03:50.517,49,83835
08-ago-2016 13:04:00.519,49,98738
08-ago-2016 13:04:10.533,50,04746
08-ago-2016 13:04:20.535,49,98788
08-ago-2016 13:04:30.535,50,07686
08-ago-2016 13:04:40.552,49,98782
08-ago-2016 13:04:50.565,50,08719
08-ago-2016 13:05:00.581,50,03733
08-ago-2016 13:05:10.583,49,8678
08-ago-2016 13:05:20.595,50,04733
08-ago-2016 13:05:30.597,50,06733
08-ago-2016 13:05:40.612,49,9974
08-ago-2016 13:05:50.626,49,86776
08-ago-2016 13:06:00.626,49,98752
08-ago-2016 13:06:10.628,49,98746
08-ago-2016 13:06:20.641,50,06749
08-ago-2016 13:06:30.642,49,99747
08-ago-2016 13:06:40.646,49,98782
08-ago-2016 13:06:50.659,49,99747
08-ago-2016 13:07:00.672,49,86779
08-ago-2016 13:07:10.673,49,98748
08-ago-2016 13:07:20.673,50,06726
08-ago-2016 13:07:30.675,49,98738
08-ago-2016 13:07:40.687,49,9974
08-ago-2016 13:07:50.689,50,08678
08-ago-2016 13:08:00.689,49,99745
08-ago-2016 13:08:10.703,49,84793
08-ago-2016 13:08:20.704,50,04746
08-ago-2016 13:08:30.705,50,06721
08-ago-2016 13:08:40.718,49,98785
08-ago-2016 13:08:50.719,49,99753

**With a too much wide time interval from a measurement and the successive one (about 10 s)!!!!!!!!**

A similar problem have been experienced in our preliminary tests (you will find the issue here https://github.com/GridProtectionAlliance/openPDC/issues/5) then successfully solved. We installed the most recent nightly build (Openpdc version 2.2.87.0 and Openhist version 2.0.360.0). Maybe the amendments implemented in the case above have not been integrated into the last versions? Please let us know as soon as you can. 

Furthermore, in general terms, how can access and extract data from the Database apart with Historian Trending tool app? 

Is it possible a remote access to the Database? If yes, how.

Many thanks,

Cosimo

StephenCWills, 2016-08-24:
Hi Cosimo,

I believe the problem you are having has to do with which archive you are accessing. In addition to the primary phasor archive (PPA), the openPDC maintains a separate archive for statistics (STAT). Statistics are sampled at 10-second intervals, which matches up with the time interval of the data you exported.

The PPA archive typically writes files to the Archive folder under the C:\Program Files\openPDC directory, and the STAT archive typically writes files to the Statistics folder under the same directory. You can change the archive that is used by the Historian Trending Tool using the blue cabinet icon in the toolbar.

[image]

Currently, there are three tools that can extract data from an openHistorian 1.0 archive.

* Historian Trending Tool for charts and CSV
* Historian Playback Utility for customizable CSV exports, but data is unaligned and transverse, meaning you get one measurement per row of data in the CSV file. This tool is considered obsolete and will be removed from future builds.
* The openPDC itself, using a LocalInputAdapter, can stream historic data as though it were real-time. The openPDC Manager can consume this data and display it on the Graph Measurements page, as can other subscription-based visualization tools, such as the GEP Subscription Tester.

In addition to these tools, data can also be accessed directly by .NET applications using the Historian API or indirectly by .NET, C++, or Java applications through the openPDC using the Subscription API. Applications using the Subscription API do have remote access to historic data.

Thanks,
Stephen

cosimopis, 2016-08-25:
Hi Stephen,

many thanks for the kind reply.

Let’s check according your hints.

Kind regards,

Cosimo

[image]<https://cloud.githubusercontent.com/assets/9203145/17946364/0de17600-6a16-11e6-87f3-4662f89aeace.png>

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

cosimopis, 2016-08-25:
Hi Stephen,

I guess that the problem still exists due to the fact that Historian Trending Tool is unable to open .d2 file of PPA (consider that we have installed OpenHistorian 2.0, the last nighlty build and not OpenHistorian 1.0 ). This means that right now the only file I’m able to open is stat_archive.d in C:\Program Files\openPDC\ STAT (which is set with the 10 s time resolution due to the reasons you explained).
How I can open ppa_XXXXXXXXX.d2 files stored by OpenHist 2.0.? Historian Trending Tool or equivalently HistorianView are no longer useful?
How I can skip the issue to access archive data?

Please let me know.

Regards,

Cosimo

[image]<https://cloud.githubusercontent.com/assets/9203145/17946364/0de17600-6a16-11e6-87f3-4662f89aeace.png>

cosimopis, 2016-08-25:
Hi Stephen,

in order to skip the issue and in integration to the previous comments to your kind answer we have tried to change database, the current one is MySql database by ensivioning the development of a home-made application for the subsequent data extraction.
The first different issue found was the one I attached below:

[cid:image001.png@01D1FEBB.5F59DFC0]

and probably due to this error no one measurement is stored, since we don’t see any of them.

Please consider this an integration to the previous raised questions, so feel free to answer just one time to all the answers in this thread.

Kind regards and thanks again,

Cosimo

[image]<https://cloud.githubusercontent.com/assets/9203145/17946364/0de17600-6a16-11e6-87f3-4662f89aeace.png>

StephenCWills, 2016-09-15:
Question was answered offline by email, so I believe this issue is closed.

For anyone checking this issue for the solution, the openHistorian 2.0 now has a tool included in the web-based user interface for querying specific time ranges to display the data on a chart.

closed 2016-09-15 by StephenCWills