GridProtectionAlliance/openPDC #27: FAIL IN CAPTURING OUTPUTSTREAM

cosimopis, 2016-09-12:
Dear all,

our partners, with which we exchange data, are unable to acquire the ouputstream we prepared for them. The situation is the following:

1) They are able to see the dataflow via PMU connection tester but due to the attribute "Data is valid" which is "False" in "Protocol Specific" their PDC discard the flow.
An example of output stream acquisition is reported below:

[image]

As you can see above "Data is valid" is False.

2) When we capture the flow directly from the unique PMU related to the data stream above the attribute "Data is valid" is True as you can see below.

[image]

By doing some tests we noted furthermore that if the PMU included in the outputstream uses directly C37118 protocol this problem vanishes. 

How we can fix this issue? 

Kind regards,

Cosimo

StephenCWills, 2016-09-12:
Hello Cosimo,

It could have something to do with this.

```
// If the expected values did not arrive for a device, we mark the data as invalid
if (m_processDataValidFlag)
{
    foreach (IDataCell cell in dataFrame.Cells)
    {
        if (!cell.AllValuesAssigned)
            cell.DataIsValid = false;
    }
}
```

And the code for AllValuesAssigned:

```
/// <summary>
/// Gets flag that determines if all values of this <see cref="DataCellBase"/> have been assigned.
/// </summary>
public virtual bool AllValuesAssigned
{
    get
    {
        return m_statusAssigned && PhasorValues.AllValuesAssigned && !FrequencyValue.IsEmpty && AnalogValues.AllValuesAssigned && DigitalValues.AllValuesAssigned;
    }
}
```

So, basically, you need to make sure that the input device is providing all the values that the openPDC is expecting to put into the output stream based on its configuration. Make sure the number of phasors, analogs, and digitals defined in the output stream match the number that exist in the input stream. Also ensure that you have mapped a status flags, frequency, and delta frequency measurement to your output stream device.

Thanks,
Stephen

ritchiecarroll, 2016-09-12:
If this is in relation to IEEE 1344, this could simply be a mapping issue, for example are "digitals" defined in the output stream that are be published from device? These can be removed from output stream.

Alternately you can just add"processDataValidFlag = false" to the connection string for output stream...

Ritchie

cosimopis, 2016-09-12:
Hi Stephen and hi Ritchie,

many thanks. I understood what is wrong. Actually if made an outputstream with one PMU which contains all its measurands the attribute  "Data is valid" is True.:

figure(1)
[image]

that matches with 

figure(2)

[image]

Let suppose that from the figure (2) I would mantain just the frequency, and one voltage magnitude and angle. How I have to set figure(1)? Secondly, how I have to consider the frequency? as a phasor?

Regards,

Cosimo

StephenCWills, 2016-09-15:
I think you can just delete the phasors and measurements you don't need. Note that you do need the delta frequency and status flags measurements. The only other caveat is that you need to fix the signal references of the phasor measurements if the numbers aren't in sequence (1, 2, 3, 4, 5, ...). For example, if you only want to keep PhasorU2, then you need to set the signal references of the corresponding measurements to PM1 and PA1.

| Label    | Type | Phase |
| -------- | ---- | ----- |
| PhasorU2 | V    | +     |

| Point ID | Signal Reference              | Source PointTag     | Historian |
| -------- | ----------------------------- | ------------------- | --------- |
| 381      | LATINA_1-FQ                   | TER_LATINA_1:F      | PPA       |
| 382      | LATINA_1-DF                   | TER_LATINA_1:DF     | PPA       |
| 383      | LATINA_1-SF                   | TER_LATINA_1:S      | PPA       |
| 386      | ~~LATINA_1-PM2~~ LATINA_1-PM1 | TER_LATINA_1-PM2:V  | PPA       |
| 387      | ~~LATINA_1-PA2~~ LATINA_1-PA1 | TER_LATINA_1-PA2:VH | PPA       |

If you want to keep PhasorU1 and PhasorI1, you need to change PhasorI1's measurements to PM2 and PA2.

| Label    | Type | Phase |
| -------- | ---- | ----- |
| PhasorU1 | V    | +     |
| PhasorI1 | I    | +     |

| Point ID | Signal Reference              | Source PointTag     | Historian |
| -------- | ----------------------------- | ------------------- | --------- |
| 381      | LATINA_1-FQ                   | TER_LATINA_1:F      | PPA       |
| 382      | LATINA_1-DF                   | TER_LATINA_1:DF     | PPA       |
| 383      | LATINA_1-SF                   | TER_LATINA_1:S      | PPA       |
| 384      | LATINA_1-PM1                  | TER_LATINA_1-PM1:V  | PPA       |
| 385      | LATINA_1-PA1                  | TER_LATINA_1-PA1:VH | PPA       |
| 388      | ~~LATINA_1-PM3~~ LATINA_1-PM2 | TER_LATINA_1-PM3:I  | PPA       |
| 389      | ~~LATINA_1-PA3~~ LATINA_1-PA2 | TER_LATINA_1-PA3:IH | PPA       |

cosimopis, 2016-09-15:
Thank you for the supprto!
I will try (meanwhile by deactivating DataisValid flag works correctly).

Regards,

Cosimo

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

ritchiecarroll, 2016-10-10:
Closing issue for now.

closed 2016-10-10 by ritchiecarroll