GridProtectionAlliance/openPDC #29: OPENPDC FREQUENT DISCONNECTION - LOG

cosimopis, 2016-09-19:
Hi,

after a period of stable operating we are experiencing frequent OpenPDC disconnections, I report in the following the most recent log:

(Inner Exception)
Date and Time:         9/19/2016 6:04:22 PM
Machine Name:          WAECNC01
Machine IP:            fe80::c9b6:2c00:d0ec:4209%21
Machine OS:            Microsoft Windows NT 6.2.9200.0
Current User ID:       PROCESSO\srv_openpdc
Current User Name:     PROCESSO\srv_openpdc
Current User Phone:    
Current User Email:    

Application Domain:    openPDC.exe
Assembly Codebase:     C:/Program Files/openPDC/openPDC.exe
Assembly Full Name:    openPDC, Version=2.2.87.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.2.87.0
Assembly Build Date:   8/3/2016 12:19:58 AM
.Net Runtime Version:  4.0.30319.42000

Exception Source:      GSF.Communication
Exception Type:        System.Net.Sockets.SocketException
Exception Message:     A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
Exception Target Site: ProcessConnect

---- Stack Trace ----
   GSF.Communication.TcpClient.ProcessConnect(connectState As ConnectState)
       openPDC.exe: N 01427

(Outer Exception)
Date and Time:         9/19/2016 6:04:22 PM
Machine Name:          WAECNC01
Machine IP:            fe80::c9b6:2c00:d0ec:4209%21
Machine OS:            Microsoft Windows NT 6.2.9200.0
Current User ID:       PROCESSO\srv_openpdc
Current User Name:     PROCESSO\srv_openpdc
Current User Phone:    
Current User Email:    

Application Domain:    openPDC.exe
Assembly Codebase:     C:/Program Files/openPDC/openPDC.exe
Assembly Full Name:    openPDC, Version=2.2.87.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.2.87.0
Assembly Build Date:   8/3/2016 12:19:58 AM
.Net Runtime Version:  4.0.30319.42000

Exception Source:      
Exception Type:        System.InvalidOperationException
Exception Message:     Connection attempt failed: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

---- Stack Trace ----

ooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooo

Furthermore today the same OpenPDC has occupied 28 GB of memory (now just 300 MB)!!!!:

Hereafter some particular  logs of the previous days:

[9/19/2016 2:54:50 AM] (Inner Exception)
(Inner Exception)
Date and Time:         9/19/2016 2:54:50 AM
Machine Name:          WAECNC01
Machine IP:            fe80::c9b6:2c00:d0ec:4209%21
Machine OS:            Microsoft Windows NT 6.2.9200.0

Application Domain:    openPDC.exe
Assembly Codebase:     C:/Program Files/openPDC/openPDC.exe
Assembly Full Name:    openPDC, Version=2.2.87.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.2.87.0
Assembly Build Date:   8/3/2016 12:19:58 AM
.Net Runtime Version:  4.0.30319.42000

Exception Source:      GSF.Communication
Exception Type:        System.Net.Sockets.SocketException
Exception Message:     An existing connection was forcibly closed by the remote host
Exception Target Site: ProcessAccept

---- Stack Trace ----
   GSF.Communication.TcpServer.ProcessAccept(acceptArgs As SocketAsyncEventArgs)
       openPDC.exe: N 01472

(Outer Exception)
Date and Time:         9/19/2016 2:54:50 AM
Machine Name:          WAECNC01
Machine IP:            fe80::c9b6:2c00:d0ec:4209%21
Machine OS:            Microsoft Windows NT 6.2.9200.0

Application Domain:    openPDC.exe
Assembly Codebase:     C:/Program Files/openPDC/openPDC.exe
Assembly Full Name:    openPDC, Version=2.2.87.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.2.87.0
Assembly Build Date:   8/3/2016 12:19:58 AM
.Net Runtime Version:  4.0.30319.42000

Exception Source:      
Exception Type:        System.Exception
Exception Message:     Unable to accept connection to client [unknown]: An existing connection was forcibly closed by the remote host

---- Stack Trace ----

(Outer Exception)
Date and Time:         9/19/2016 2:54:50 AM
Machine Name:          WAECNC01
Machine IP:            fe80::c9b6:2c00:d0ec:4209%21
Machine OS:            Microsoft Windows NT 6.2.9200.0

Application Domain:    openPDC.exe
Assembly Codebase:     C:/Program Files/openPDC/openPDC.exe
Assembly Full Name:    openPDC, Version=2.2.87.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.2.87.0
Assembly Build Date:   8/3/2016 12:19:58 AM
.Net Runtime Version:  4.0.30319.42000

Exception Source:      
Exception Type:        System.InvalidOperationException
Exception Message:     Socket exception occurred while client attempting to connect to command channel: Unable to accept connection to client [unknown]: An existing connection was forcibly closed by the remote host

ooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooo

Could you provide some indications at this regard?

Please let us known

Regards,

Cosimo

StephenCWills, 2016-09-23:
> A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

This error typically indicates that there is a firewall blocking packets between the openPDC and the device it is connecting to. In other words, as part of the TCP three-way handshake, openPDC sent a SYN packet to the remote host, but the remote host did not respond with an ACK packet for the duration of the timeout period. The timeout is typically something like 15 or 30 seconds, so while it technically can also occur in cases when the SYN or ACK packets were somehow dropped due to an error on the network or when the network has severe performance issues, these cases are much less common.

---

> An existing connection was forcibly closed by the remote host

This error occurs when the remote host sends an RST packet. These are typically used to indicate that an error has occurred in the application on the remote host that is connecting to the openPDC, and that the remote host has aborted the connection without the opportunity for issuing a graceful shutdown sequence.

---

I'm not sure of what exactly would be causing high memory usage in the openPDC, but it's not entirely unusual to see that in cases where a consumer of real-time data within the system is spending too much time processing. Since the data never stops coming from the upstream device, that data simply builds up in the queues maintained by the openPDC's internal measurement routing engine and this causes memory to rise uncontrollably. It could be that there was a network performance issue that caused delays when attempting to send data from the openPDC to external systems and that the system ran out of memory before it was allowed to recover.

Thanks,
Stephen

cosimopis, 2016-09-26:
Dear Stephen,

thanks for the reply. Do you have some hints about an efficient memory management (e.g. PMUs sharing on different servers) ? .... and hence related OpenPDC/OpenHIST configuration

Please let us know

Cosimo

StephenCWills, 2016-10-06:
Cosimo,

1) For adapters that create network sockets, there should be a parameter that allows you to specify the maximum number of messages to build up before dumping data from the queue (MaxSendQueueSize=500000 is the default). Historian adapters and output adapters have a similar feature  which is controlled by some global threshold settings in the config file (MeasurementWarningThreshold and MeasurementDumpingThreshold in the /configuration/categorizedSettings/thresholdSettings section of openPDC.exe.config). If the measurements are instead building up in the routing layer, there is currently no way to relieve the pressure besides restarting the system. That feature is on our to-do list.

2) Other than blindly lowering thresholds or improving hardware, it's difficult for me to say what might help to reduce the memory usage on your system. Typically this is caused by some kind of bottleneck, but it could also be caused by an error or a particularly inefficient piece of code. There may be a virus scanner or backup process running on the system that is using resources on the machine.

I have to agree that increasing RAM will only delay the inevitable. If you can, the most valuable thing you could do at this point is to identify the cause of the buildup. Based on some of the other discussions you've had with Ritchie, it sounds like you're on the right track.

Thanks,
Stephen

cosimopis, 2016-10-06:
Hi Stephen,

I upgraded openHIST with last nightlybuild and changed the storage disk technology. Now it seems that the memory is under control. Anyway these hints are very valuable and I will soon investigate the effect in managing the queues according the config. file of openPDC.

Thank you for the help.

Kind regards,

Cosimo

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

ritchiecarroll, 2016-10-10:
Closing issue for now.

closed 2016-10-10 by ritchiecarroll