GridProtectionAlliance/openPDC #32: OPENPDC STAT FILE

cosimopis, 2016-09-30:
Hi guys,

in view of a reduction of the data written by OpePDC on our server since we don't need stat file (stat_archive.d), is there a manner to avoid their writing?

If not, Can I install Opendc (es. C:\) in a directory and impose that it writes file in a different path (es. E:\)?

For instance by modifying the file OpenPDC.exe.config in the related rows?

Kind regards

ritchiecarroll, 2016-09-30:
I would not recommend disabling the statistics archive as you may want to turn daily data quality reports on in the future that will built upon this data. You can, however, move the files to a different folder and/or drive. This is usually as simple as changing the paths to locations related to "stat" files in the openPDC.exe.config file, see XML snippets below.

It is generally safest to use the "XML Configuration Editor" for changes to the configuration file;  this tool will even automatically restart the service for you to apply changes. If you make changes to the XML manually, you must make sure the openPDC is not running before you make save any updates to the file - and - I would always recommend backing the file up if you for manual changes (note that using the XML Configuration Editor also makes a backup for you). 

*Important!* You should setup a new folder with needed rights on the E:\  _before_ you make changes to the configuration file (see notes below XML snippet).
```
    <statMetadataFile>
      <add name="FileName" value="e:\Archive\Statistics\stat_dbase.dat" description="Name of the statistics meta-data file including its path."
        encrypted="false" />
    </statMetadataFile>
    <statStateFile>
      <add name="FileName" value="e:\Archive\Statistics\stat_startup.dat" description="Name of the statistics state file including its path."
        encrypted="false" />
    </statStateFile>
    <statIntercomFile>
      <add name="FileName" value="e:\Archive\Statistics\scratch.dat" description="Name of the statistics intercom file including its path."
        encrypted="false" />
    </statIntercomFile>
    <statArchiveFile>
      <add name="FileName" value="e:\Archive\Statistics\stat_archive.d" description="Name of the statistics working archive file including its path."
        encrypted="false" />
    </statArchiveFile>
```
The openPDC, by default, runs as the "NT SERVICE\openPDC" user which has security rights to very little, so the service will certainly not be able to write files into a new folder on a different drive until explicitly granted rights to do so.

For example, if you created a new folder on the E:\ drive called "Archive", you would need to right click on the folder an select "Properties" then click on the "Security" tab. Click the "Edit" button" then click "Add". Type in "NT SERVICE\openPDC" (without quotes) and the click the "Locations..." button then select the root computer node and click the "OK" button. You show now be back on the "Select Users or Groups" screen and you can click "OK".

[image]

The user "openPDC" should now be in the list. Select the openPDC user and make sure rights for "Modify" is checked, then click "OK" and "OK" again. Now the openPDC should have the needed rights to write new files into this folder.

Before restarting the openPDC with updated configuration changes you will want to *move* all of the existing files statistic files into the new folder, i.e., move "C:\Program Files\openPDC\Statistics\*.*" to "E:\Archive\Statistics\".

Hope that helps!
Ritchie

closed 2016-09-30 by ritchiecarroll

cosimopis, 2016-10-03:
Hi Ritchie,

thank you for all the clarifications. The problem is that our hadware right now is compliant to the design requirements you suggested in (https://github.com/GridProtectionAlliance/openPDC/issues/7). To be honest the numeber of PMUs il lower than 70. Unfortunately  we are experiencing some memory issues: OpenPDC increase gradually its own memory occupation by stopping to acquire data from the field and losting all the data which is crucial for understanding out of services occurred on the network. 

To reduce the loading of the server our simple idea is to reduce the number of files written by the server in the working directories. In this sense the principal paths on which act are:

1)" C:\Program Files\openPDC\Statistics" --> stat data
2) "C:\Program Files\openPDC\ConfigurationCache\" --> xml config file I don't know why frequent updated.

If I limit the amount of written files I will probably of better control on the memory resource. Do you have  
any suggestion to investigate and try to fix the MEMORY LEAK issue? This should be appreciated.  

Kind regards,

Cosimo

cosimopis, 2016-10-03:
In addition, I know that stat data are useful for report monitoring but is there a manner to avoid stat files writing?

Please let me know

ritchiecarroll, 2016-10-03:
I am a little confused, writing stat filed relates to disk space usage not memory utilization. Is the disk filling up or is there a memory issue?

cosimopis, 2016-10-03:
We are experiencing a memory issue (memory leaks). The main responsible seems to be openPDC which increases the “threads” and “handles” during the time (see task manager options). By monitoring the task manager we noted that some heavy files are written by openpdc in Statistic folder such as “stat_startup.dat”. In order to limit the use of memory we are thinking to reduce the info we don’t need. Part of them are are stat file. What do you suggest?

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

ritchiecarroll, 2016-10-03:
To disable the Statistics historian you can simply go to the openPDC Manager and navigate to Outputs / Historians, select the STAT historian row and uncheck "Enabled". Then click "Save" and "Initialize" - this should disable writing to the Statistics archive.

I will speculate that perhaps all the file activity you are seeing is related to the constant configuration changes - this would cause unnecessary churn for the statistics historian. Perhaps your best course of action is to determine "what" is changing in the configuration, if we can figure that out perhaps we can assist you in reducing the load.