GridProtectionAlliance/openPDC #33: \openPDC\ConfigurationCache\"

cosimopis, 2016-09-30:
Hi,

I noted that OpenPDC try to update with high frequency the PMU config file in the path above. In view of reducing the amount of written file, is it possibile to set a frequency of refresh for them?

Typically once set up the configuration don't change.

Please let us know.

Regards

ritchiecarroll, 2016-09-30:
Which files are being update frequently? I agree that for static configuration, this should change infrequently. This could be indicative of a configuration issue.

cosimopis, 2016-10-01:
The ones in the path C:\Program Files\openPDC\ConfigurationCache\" which are xml files with the name of the devices (I didn't open them but I think there are the config info of each PMU).

Regard

Inviato tramite OWA per iPhone

A: GridProtectionAlliance/openPDC
Cc: Pisani Cosimo (Terna); Author

Questo messaggio (allegati eventuali compresi) contiene informazioni del Gruppo Terna che devono essere utilizzate esclusivamente dai destinatari in relazione alle finalità per le quali sono state inviate. È vietata qualsiasi forma di divulgazione senza l’esplicito consenso di Terna. Qualora questa e-mail sia stata ricevuta per errore, si prega di provvedere alla distruzione dell’intero messaggio e di informare tempestivamente il mittente. Grazie.

This email (and any files transmitted with it) includes Terna Group information and is intended solely for the use of the recipients to whom it is addressed. Please note that disclosing the contents of this email is strictly prohibited without Terna’s express consent. Please notify the sender immediately if you have received this e-mail by mistake and delete this e-mail from your system. Thank you.

ritchiecarroll, 2016-10-01:
These normally only get updated when the configuration of the source device changes. Can you grab a copy of the files, then after the change grab another copy an compare the differences - you can use a tool such as BeyondCompare http://www.scootersoftware.com/download.php to look at what data in the XML files is changing.

ritchiecarroll, 2016-10-06:
Have you determined the cause of frequent configuration changes - this could be a clue?

cosimopis, 2016-10-06:
Hi Ritchie,

unfortunately I didn’t discover why yet. But I upgraded openHIST with last nightlybuild and changed the storage disk technology. Now it seems that the memory is under control.

Thak you for the help. If I will have news about the issue I will upgrade the thread.

Kind regards,

Cosimo

ritchiecarroll, 2016-10-10:
Closing for now.

closed 2016-10-10 by ritchiecarroll