GridProtectionAlliance/openPDC #35: openPDC Manager config setting doesn't stick

mlbeauchemin, 2016-10-11:
We have attempted to change the FileSize setting in the openPDC.exe.config file under 'statArchiveFile' from 2 to 100 a couple of times, but it keeps resetting back to 2. Is this a bug?

StephenCWills, 2016-10-11:
The openPDC service writes to the config file periodically, and the settings are cached in memory. The most reliable way to update the config file is to do it while the service is not running.

mlbeauchemin, 2016-10-11:
It seems odd because that's the only setting we've run into so far with this particular quirk, but stopping the service beforehand seems to have worked to get it to change. Thanks.

StephenCWills, 2016-10-11:
Actually, you may want to double-check that. I wasn't paying enough attention to the config file setting you were adjusting. That setting is tuned automatically by the OptimizeLocalHistorianSettings data operation, so it will be modified by the service on every ReloadConfig.

https://github.com/GridProtectionAlliance/gsf/blob/master/Source/Libraries/Adapters/HistorianAdapters/LocalOutputAdapter.cs#L994

If you would like to change it manually, you would need to prevent that data operation from executing. There is a record in the DataOperation table that controls whether that OptimizeLocalHistorianSettings method runs during ReloadConfig. If you set the Enabled flag to 0, you can keep it from stomping over your settings.

```
UPDATE DataOperation SET Enabled = 0 WHERE MethodName = 'OptimizeLocalHistorianSettings'
```

You may want to read the comments in the code I linked above before making a final decision. They explain the rationale behind how the STAT file size is calculated based on the number of statistics defined in your system.

closed 2016-11-03 by ritchiecarroll