GridProtectionAlliance/openPDC #56: Some MySQL script errors when I execute ConfigurationSetupUtility

siaDaniel, 2016-11-29:
Hi everyone,
 
I run the ConfigurationSetupUtility and choose MySQL as the configuration type. Enter the host, database, username, password. And pass the Test Connection. After some next options, it starts to execute openPDC.sql script. But an error occurs. The log is as follows:

> An unexpected exception has occurred in ConfigurationSetupUtility. This may be due to an inconsistent system state or a programming error.
> 
> Error details: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Virtual
> FROM OutputStreamDevice OSD' at line 8

Then I run openPDC\Database Scripts\MySQL\openPDC.sql. The error is as below. Should I run other script files first? Or the script has some errors itself? How can I solve this? Thanks in advance.

> [Err] 1064 - You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Virtual
> FROM OutputStreamDevice OSD;
>                       
> CREATE VIEW Phaso' at line 8
> [Err] ALTER TABLE ApplicationRoleSecurityGroup ADD CONSTRAINT FK_applicationrolesecuritygroup_securitygroup FOREIGN KEY (SecurityGroupID) REFERENCES securitygroup (ID) ON DELETE CASCADE ON UPDATE CASCADE;
> 
> ALTER TABLE UserAccount ADD CONSTRAINT FK_useraccount FOREIGN KEY (DefaultNodeID) REFERENCES node (ID) ON DELETE CASCADE ON UPDATE CASCADE;
> 
> ALTER TABLE ApplicationRole ADD CONSTRAINT FK_applicationrole FOREIGN KEY (NodeID) REFERENCES node (ID) ON DELETE CASCADE ON UPDATE CASCADE;
> 
> ALTER TABLE ApplicationRoleUserAccount ADD CONSTRAINT FK_applicationroleuseraccount_useraccount FOREIGN KEY (UserAccountID) REFERENCES useraccount (ID) ON DELETE CASCADE ON UPDATE CASCADE;
> 
> ALTER TABLE ApplicationRoleUserAccount ADD CONSTRAINT FK_applicationroleuseraccount_applicationrole FOREIGN KEY (ApplicationRoleID) REFERENCES applicationrole (ID) ON DELETE CASCADE ON UPDATE CASCADE;
> 
> ALTER TABLE SecurityGroupUserAccount ADD CONSTRAINT FK_securitygroupuseraccount_useraccount FOREIGN KEY (UserAccountID) REFERENCES useraccount (ID) ON DELETE CASCADE ON UPDATE CASCADE;
> 
> ALTER TABLE SecurityGroupUserAccount ADD CONSTRAINT FK_securitygroupuseraccount_securitygroup FOREIGN KEY (SecurityGroupID) REFERENCES securitygroup (ID) ON DELETE CASCADE ON UPDATE CASCADE;
> 
> CREATE VIEW NodeInfo
> AS
> SELECT Node.ID AS NodeID, Node.Name, Company.Name AS CompanyName, Node.Longitude, Node.Latitude, Node.Description,
>     Node.ImagePath, Node.Settings, Node.MenuType, Node.MenuData, Node.Master, Node.Enabled
> FROM Node LEFT OUTER JOIN Company ON Node.CompanyID = Company.ID;
> 
> CREATE VIEW RuntimeOutputStreamMeasurement
> AS
> SELECT OutputStreamMeasurement.NodeID, Runtime.ID AS AdapterID, Historian.Acronym AS Historian, 
>     OutputStreamMeasurement.PointID, OutputStreamMeasurement.SignalReference
> FROM OutputStreamMeasurement LEFT OUTER JOIN
>     Historian ON OutputStreamMeasurement.HistorianID = Historian.ID LEFT OUTER JOIN
>     Runtime ON OutputStreamMeasurement.AdapterID = Runtime.SourceID AND Runtime.SourceTable = N'OutputStream'
> ORDER BY OutputStreamMeasurement.HistorianID, OutputStreamMeasurement.PointID;
> 
> CREATE VIEW RuntimeHistorian
> AS
> SELECT Historian.NodeID, Runtime.ID, Historian.Acronym AS AdapterName,
>     COALESCE(NULLIF(TRIM(Historian.AssemblyName), ''), N'HistorianAdapters.dll') AS AssemblyName, 
>     COALESCE(NULLIF(TRIM(Historian.TypeName), ''), IF(IsLocal = 1, N'HistorianAdapters.LocalOutputAdapter', N'HistorianAdapters.RemoteOutputAdapter')) AS TypeName, 
>     CONCAT_WS(';', Historian.ConnectionString, CONCAT(N'instanceName=', Historian.Acronym), CONCAT(N'sourceids=', Historian.Acronym),
>     CONCAT(N'measurementReportingInterval=', CAST(Historian.MeasurementReportingInterval AS CHAR))) AS ConnectionString
> FROM Historian LEFT OUTER JOIN
>     Runtime ON Historian.ID = Runtime.SourceID AND Runtime.SourceTable = N'Historian'
> WHERE (Historian.Enabled <> 0)
> ORDER BY Historian.LoadOrder;
> 
> CREATE VIEW RuntimeDevice
> AS
> SELECT Device.NodeID, Runtime.ID, Device.Acronym AS AdapterName, Protocol.AssemblyName, Protocol.TypeName,
>     CONCAT_WS(';', Device.ConnectionString, CONCAT(N'isConcentrator=', CAST(Device.IsConcentrator AS CHAR)),
>     CONCAT(N'accessID=', CAST(Device.AccessID AS CHAR)),
>     IF(Device.TimeZone IS NULL, N'', CONCAT(N'timeZone=', Device.TimeZone)),
>     CONCAT(N'timeAdjustmentTicks=', CAST(Device.TimeAdjustmentTicks AS CHAR)),
>     IF(Protocol.Acronym IS NULL, N'', CONCAT(N'phasorProtocol=', Protocol.Acronym)),
>     CONCAT(N'dataLossInterval=', CAST(Device.DataLossInterval AS CHAR)),
>     CONCAT(N'allowedParsingExceptions=', CAST(Device.AllowedParsingExceptions AS CHAR)),
>     CONCAT(N'parsingExceptionWindow=', CAST(Device.ParsingExceptionWindow AS CHAR)),
>     CONCAT(N'delayedConnectionInterval=', CAST(Device.DelayedConnectionInterval AS CHAR)),
>     CONCAT(N'allowUseOfCachedConfiguration=', CAST(Device.AllowUseOfCachedConfiguration AS CHAR)),
>     CONCAT(N'autoStartDataParsingSequence=', CAST(Device.AutoStartDataParsingSequence AS CHAR)),
>     CONCAT(N'skipDisableRealTimeData=', CAST(Device.SkipDisableRealTi
> [Msg] Finished - Unsuccessfully
> --------------------------------------------------

StephenCWills, 2016-11-29:
This is an issue that was introduced into the openPDC.sql script by MySQL version 5.7.6 when the VIRTUAL keyword was introduced. Here is a link to the fix.
https://github.com/GridProtectionAlliance/gsf/commit/0a980abe74469b1b052573455019c78f53d3368d

In your case, you will either need to install the latest version of openPDC (actually, any of the releases on GitHub should work as this fix was included in version 2.2) or fix the script manually. The script can be found in the `C:\Program Files\openPDC\Database Scripts` directory.

closed 2016-11-29 by StephenCWills

siaDaniel, 2016-11-30:
Thanks Stephen.