GridProtectionAlliance/openPDC #88: Help with Linux Install

avs009, 2016-12-29:
I need some help to execute openPDC in Debian Jessie:

1- The original script in wiki instructions ([Deploying the openPDC on POSIX Platforms](https://gpags.sharepoint.com/TeamSite/_layouts/15/WopiFrame.aspx?guestaccesstoken=ADyQzHPxsfTh9qs4glPelL78SoBA1pTJV1%2fWy6b0ct4%3d&docid=08819043371f24a089e4924e86525dd69&action=view) not working with last release 2.3

2- After modified the script i was able to compile and install, but things not are perfect, the openPDC not run and return this message only:
`# mono /opt/openPDC/openPDC.exe -RunAsConsole`
`Shell hosted service exception: Object reference not set to an instance of an object`

Versions:

openPDC 2.3 Source from git
no GSF from source (suposed it is included on openPDC src)
Mono JIT compiler version 4.6.2 from mono-project for wheezy
Debian Jessie 8

[openPDC.exe.config.zip](https://github.com/GridProtectionAlliance/openPDC/files/677475/openPDC.exe.zip)

ajstadlin, 2016-12-29:
I was able to get openPDC running on Raspberry Pi Model 3 and Zero using the following procedures:
[Running_openPDC_on_a_Raspberry_Pi.files](https://github.com/GridProtectionAlliance/openPDC/tree/master/Source/Documentation/wiki/Running_openPDC_on_a_Raspberry_Pi.files)

While developing the Raspberry Pi procedures, I did pretty practiced the same procedure in Debian Jessie network minimal install.  However, it looks like my Raspbian tests were with openPDC 2.2.70.

Did you get the install-openpdc.sh script from [https://www.gridprotectionalliance.org/Products/openPDC/Scripts/](https://www.gridprotectionalliance.org/Products/openPDC/Scripts/) ?

//aj

avs009, 2017-01-02:
Hi, 

Yes i get script from `https://www.gridprotectionalliance.org/Products/openPDC/Scripts/`

I tested now with original script and nightly build as you suggested, and both works sucessfull.

I modified the script to get from git sources acording a branch you put in command line or open pdc releases, and option re-build GSF.

My motivation is for use with last stable release in linux, but i think GSF dependency broken my compilation from source.

At this moment i write a version with your requirements in attachements, is not fully tested.
Not working with last source of GSF, occur compile error.
In help has some examples for use.

[install-openPDC.zip](https://github.com/GridProtectionAlliance/openPDC/files/681344/install-openPDC.zip)

ajstadlin, 2017-01-22:
Thank you for the revised install-openPDC.sh script.  I like it!  It will take me some time to test the variations - and make up an exercise for it.

ritchiecarroll, 2017-01-28:
Is this still an issue or can this be closed?

ajstadlin, 2017-01-30:
I think this can be closed.  I tried the install-openPDC.sh script with several variations and it seems to work, but I haven't had a chance to run documented tests yet.  Can this script be included in the openPDC project so others use it and provide feedback?

ajstadlin, 2017-01-30:
This is what the script **-h** looks like:

```
openPDC installation script
    Options:
        -h: Show help
        -s: Git base address (defaults to https://github.com/GridProtectionAlliance)
        -n: Use git for source (defaults to NO, only download releases)
        -v: Release version for download (default 2.3)
        -p: Preserve source code, (not download source again, as default)
        -d: Install destination (defaults to /opt/openPDC)
        -c: Download and compile GSF from source (defaults to NO, use GSF included in openPDC)
        -f: Force to selected branchs configured below (overwrite local changes)
        -g: Branch to obtain Grid Solutions Framework source from git (defaults to master)
        -o: Branch to obtain openPDC source from git (defaults to master)
        -u: Uninstall service
    Examples of use:
        install-openPDC.sh -h
        install-openPDC.sh -v 2.3
        install-openPDC.sh -n
        install-openPDC.sh -s "https://github.com/GridProtectionAlliance" -n
        install-openPDC.sh -n -o "v2.3"
        install-openPDC.sh -n -c -g "master" -o "v2.3"
        install-openPDC.sh -s "https://github.com/GridProtectionAlliance" -n -g "master" -o "v2.3"
        install-openPDC.sh -u
```

ajstadlin, 2017-01-30:
This modified install-openPDC.sh script could be renamed with an alternative name if included in the openPDC project.

avs009, 2017-01-30:
@ritchiecarroll, yes this problem can be closed, the problem related here occurred because I not compile GSF for Linux.

@ajstadlin Thanks for using my script, if you need any help with options available call me.

ritchiecarroll, 2017-01-30:
Feel free to include script in openPDC AJ - thanks! Looks useful - thanks for the add...

closed 2017-01-30 by ritchiecarroll