GridProtectionAlliance/openPDC #94: Trouble Subscribing openHistorian to openPDC

sbasumal, 2017-07-02:
Hi,

I am trying to archive real time streaming data from two SEL 421 relay using openHistorian and I face the following issue: I **cannot view real time data in openHistorian after subcription, but the same can be viewed in openPDC**. I have used the same connection string, same port number, same acronym and everything for both the software.
[image]

Here are the settings:
[image]

While subscribing, i left the host name empty as I am using a serial connection:
[image]

 Here is my COM3 port settings:
[image]

StephenCWills, 2017-07-03:
Hi sbasumal,

1. Serial ports are exclusive resources, meaning that two separate pieces of software cannot use the same serial port at the same time. If you need to get the data into both applications, you can connect openPDC to the SEL-421 devices and then connect openHistorian to openPDC. The connection between openHistorian and openPDC can be established rather easily by using openHistorian Manager to create an internal subscription to port 6165.

2. On the `Device Configuration` page of the openHistorian Manager, there is a combo box labeled `Historian`. Using that combo box, you can select the destination archive for the data coming from that device. In the screenshots you provided before editing this issue, it looks like you had `PPA` selected as the destination archive, so the data that was plotted in your latest screenshot should already have been saved in the PPA archive. You can use the `Trend / Export Archived Data` page of the openHistorian Web Manager to plot or export archived data. That should allow you to verify whether the data has been successfully archived.

Thanks,
Stephen

sbasumal, 2017-07-03:
Thanks, it worked. I re-installed the software and things worked out fine.

closed 2017-07-03 by StephenCWills