GridProtectionAlliance/openPDC #95: about protocols

Tibalt, 2017-08-16:
Hi, 

I browsed the code for some time and looked through the issues but I haven't found the code for protocols  such as 61850-90-5 and 37118. 

Are the protocols included in openPDC ? If so, can anybody help me to find the code?   

Thanks in advance.

Regards
Tibalt

Tibalt, 2017-08-16:
seems I could answer my own question:
openPDC relies on gsf from GPA. The link is https://github.com/GridProtectionAlliance/gsf
The documentation only say the gsf could be deployed on windows 7 or newer. It seems now gsf could be deployed in Linux and other POSIX system? Otherwise openPDC cannot run on Linux.

Regards
Tibalt

ritchiecarroll, 2017-08-16:
Correct, and openPDC runs on Linux just fine:
https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/Running_openPDC_on_Linux_and_Mac.md

Thanks,
Ritchie

closed 2017-08-16 by ritchiecarroll