GridProtectionAlliance/openPDC #96: installation on rasp3

Tibalt, 2017-08-17:
I followed the [link](https://github.com/GridProtectionAlliance/openPDC/blob/26afda6966653c70fde3aa3417660aaa3811b21e/Source/Documentation/wiki/Running_openPDC_on_a_Raspberry_Pi.files/Example-openPDC_on_RaspberryPi-3B_Raspbian_Jessie_Lite.pdf) to install openPDC on rasp 3 and unlucky. 
It seem mono is good because I see below:

tibalt@raspberrypi:~/GPA $ xbuild /version
XBuild Engine Version 14.0
Mono, Version 4.4.2.0
Copyright (C) 2005-2013 Various Mono authors

And the build log is attached and it is hard for me to find any clue to fix the errors.

Please help and many thanks.

Regards,
Tibalt
[install-openPDC.log.txt](https://github.com/GridProtectionAlliance/openPDC/files/1230863/install-openPDC.log.txt)

ritchiecarroll, 2017-08-17:
What Linux version are you targeting? Looks like the mono version there doesn't support UI features.

Have you tried our pre-compiled Rasbian image?
https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/Running_openPDC_on_a_Raspberry_Pi.md

Thanks,
Ritchie

Tibalt, 2017-08-18:
The uname -a says:

_tibalt@raspberrypi:~/GPA $ uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux_

As the image is for rasp 1 and rasp 2 and I am using rasp 3, I haven't tried it. 
What exactly the mono version is good?  I found this in 
[image]

http://www.gridprotectionalliance.org/docs/products/openPDC/openPDConPOSIX.pdf

Tibalt, 2017-08-18:
by the way, the rasp 3 is not running a desktop version Linux.

ajstadlin, 2017-08-18:
Are you using the **install-openPDC.sh** script from the Products scripts: [http://www.gridprotectionalliance.org/Products/openPDC/Scripts/install-openPDC.sh](http://www.gridprotectionalliance.org/Products/openPDC/Scripts/install-openPDC.sh) 

Another script was proposed in [Issue #88 Help with Linux Install](https://github.com/GridProtectionAlliance/openPDC/issues/88)

The script is found in [install-openPDC.zip](https://github.com/GridProtectionAlliance/openPDC/files/681344/install-openPDC.zip)

I noticed that the current `~/Products/openPDC/Scripts/install-openPDC.sh` is not the same, so possibly give the #88 script a try.

//aj

Tibalt, 2017-08-18:
I get it through _wget http://www.gridprotectionalliance.org/Products/openPDC/Scripts/install-openPDC.sh_ 
I tried openpdc 2.2 stable by modifying openPDC.sh, still no luck. 
I will give #88 a try.

Tibalt, 2017-08-18:
The #88 not work for me. 
The log is attached. 
[install-openPDC.txt](https://github.com/GridProtectionAlliance/openPDC/files/1233841/install-openPDC.txt)
This time the failure comes quicker than the 
wget http://www.gridprotectionalliance.org/Products/openPDC/Scripts/install-openPDC.sh

Tibalt, 2017-09-14:
I tried the openPDC image for rasp 2 and it worked perfectly. ：）

Then I tried again to make the openPDC running on the latest raspbian and mono. 

The OS installed by me which failed the compilation is of lite version while the openPDC image has desktop.  So that I gave the "full" image a try. This time the compilation is well passed! ( I did not fully followed the instructions because the officially recommended way to install mono in [http://www.mono-project.com/download/#download-lin-raspbian](url) takes much less time. ) The mono version I installed is 

> mono --version
Mono JIT compiler version 5.2.0.215 (tarball Mon Aug 14 16:59:51 UTC 2017)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       normal
        Notifications: epoll
        Architecture:  armel,vfp+hard
        Disabled:      none
        Misc:          softdebug
        LLVM:          supported, not enabled.
        GC:            sgen (concurrent by default)

The rasp image is 

> i@openpdc:/opt/openPDC $ uname -a
Linux openpdc 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
pi@openpdc:/opt/openPDC $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
[install-openPDC.log.txt](https://github.com/GridProtectionAlliance/openPDC/files/1302087/install-openPDC.log.txt)

But there are somthing wrong when tried to start the openpdc by:

> sudo mono ./openPDC.exe -RunAsConsole
Shell hosted service exception: Object reference not set to an instance of an object.

I saw  two errors in /var/log/messages:

> Sep

 13 09:06:32 openpdc lightdm[466]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files

Sep 13 09:06:34 openpdc lightdm[534]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files

I think it is very close to run it successfully. Any hint?

Many thanks!

Regards
Tibalt

ajstadlin, 2017-09-21:
Thank you for your perseverance!

I am personally not able to work on this at this time and do not have any hints. However, I'm monitoring this issue and hope it eventually gets worked out. Raspbian Stretch / Debian 9 is really improved over the Jessie versions; and the Raspberry Pi is a great inexpensive device for experimenting and learning.

//aj

ritchiecarroll, 2022-04-01:
Try the latest release:
https://github.com/GridProtectionAlliance/openPDC/releases/tag/v2.9.148

closed 2022-04-01 by ritchiecarroll