GridProtectionAlliance/openPDC #101: Custom Point Tag Naming

logiclabsrl, 2017-12-12:
Hi,
I'm working with OpenPDC release 2.4.120.0. I'd like to modify default tag naming expression:

{CompanyAcronym}_{DeviceAcronym}[?{SignalType.Source}=Phasor[-{SignalType.Suffix}{SignalIndex}]]:{VendorAcronym}{SignalType.Abbreviation}[?{SignalType.Source}!=Phasor[?{SignalIndex}!=-1[{SignalIndex}]]]

I create the desired expression and I made the change on both the configuration files (openPDC.exe.config and openPDCManager.exe.Config) without obtaining the desired result. Even simple modifications of the expression (I changed a "_" with "-") are not applied to the tag name.

Is it necessary to make a change to other parameters? Is it necessary to perform a special reset and / or initialization procedure?

Any help would be really appreciated.

Thanks

ritchiecarroll, 2017-12-12:
#1 **_Back up your configuration database before doing this work in case things don't go well - that way you can fall back to an operational starting point. Note that backing a database is a database specific operation, for example: in SQL Server you would open the Microsoft SQL Server Management Studio, right-click on the target database and select `Tasks > Back Up...` - for SQLite you would only need to make a separate copy of the `.db` file._**

To force application of the new expression to existing tag names overwriting them with the new expression you will need to add `renameAllPointTags=true` to the `Arguments` field of the `Phasor Data Source Validation` data operation. The record can be found in the `DataOpeperations` table in your primary configuration.

For example, if you are using SQL Server, you can open the Microsoft SQL Server Management Studio, browse to your openPDC data, then right-click on the `dbo.DataOperation` table and select ` Edit Top 200 Rows` - then you can paste `renameAllPointTags=true` into the `Arguments` field and click off of the record to apply the changes. Same kind of thing with other databases, just use a database browsing tool to do the same type of steps, or just use a SQL expression like the following:
```
UPDATE DataOperation SET Arguments='renameAllPointTags=true' WHERE MethodName='PhasorDataSourceValidation'
```

Once the record has been updated, you need only issue a `ReloadConfig` command from the openPDC console or just restart the service - either way, the configuration will be reloaded and the new expression will be applied to all the points in the system.

Note that the arguments flag will be automatically cleared from the record once the rename is complete so that it only applies the rename operation once. You will need to manually add the flag back if you need to apply the rename operation again.

Hope that helps, good luck!

Thanks,
Ritchie

ritchiecarroll, 2017-12-12:
@chefsteph9  - please post this as an FAQ to the openPDC Wiki somewhere... Thanks!

logiclabsrl, 2017-12-12:
@ritchiecarroll - It works! Thank you very much

I have only a problem with field {PhasorLabel}. If I use it inside tag name expression, it is not resolved properly in the tag name.

For example, with this test expression
\[{DeviceAcronym}\] [?{SignalType.Source}=Phasor[{PhasorLabel}-CH{SignalIndex}-{SignalType.Abbreviation}{Phase}]]:{SignalType.LongAcronym}[?{SignalType.Source}!=Phasor[?{SignalIndex}!=-1[{SignalIndex}]]]

The result for device "SPARE1" is
[SPARE1] {PhasorLabel}-CH1-VA:VoltageMagnitude

Considering the Phasor table of unit SPARE1 as shown in the image below, what I would have expected is a tag name like this
[SPARE1] CHANNEL_1-CH1-VA:VoltageMagnitude

Where is the mistake?
Thanks,
Daniele

[image]

ritchiecarroll, 2017-12-12:
Looks like it's not properly replacing the `{PhasorLabel}` tag for some reason. Note that brackets used around device acronym are reserved characters, perhaps you should escape these and try again:

\[{DeviceAcronym}\] [?{SignalType.Source}=Phasor[{PhasorLabel}-CH{SignalIndex}-{SignalType.Abbreviation}{Phase}]]:{SignalType.LongAcronym}[?{SignalType.Source}!=Phasor[?{SignalIndex}!=-1[{SignalIndex}]]]

ritchiecarroll, 2017-12-12:
FYI, more complex C# like expressions are also support using the `eval{}` function:
```
eval{'{CompanyAcronym}'.Substring(0,3)}_{DeviceAcronym}eval{'[?{SignalType.Source}=Phasor[-{SignalType.Suffix}]]'.Length}
```

ritchiecarroll, 2017-12-12:
See paragraph that begins with "Advanced expressions" topic in *Remarks*:
https://www.gridprotectionalliance.org/NightlyBuilds/GridSolutionsFramework/Help/html/M_GSF_Parsing_TemplatedExpressionParser_Execute.htm

logiclabsrl, 2017-12-14:
Thanks for your support.

I have done other tests these days.
- I tried to use the single entry {PhasorLabel} as the expression and the result was that all the measurements had the same name tag {PhasorLabel} not replaced by the real phasor name.
- I tried to use the eval{} function as suggested, but it seems that the use of this type of function is not fully supported since the measurements maintain the name in the compliant to previous format defined in the expression (tag name is not updated to new expression format).
- I tried to use the expression you suggested in the comment above but this also gave a negative result.
- I tried to use the {PhasorLabel} entry on an openpdc instance that works on a different system but with the same result: the entry is not correctly replaced in the tag name.

Any other suggestion?
Thanks

ritchiecarroll, 2018-01-29:
`{PhasorLabel}` will only substitute on measurements with a signal type of `VPHA`, `VPHM`, `IPHA` and `IPHM` - measurement types of any other signal type will have a blank `{PhasorLabel}` property. Also, the signal reference index, i.e., the last numeric value in the `{SignalReference}` field for measurements with a signal type of `VPHA`, `VPHM`, `IPHA` and `IPHM`, must match one of the indexes in the `Phasor` table where the `Phasor.DeviceID` matches `Measurement.DeviceID` - if not, again the `{PhasorLabel}` property will be blank.

ritchiecarroll, 2019-07-15:
I have improved the operation of the custom point tag naming convention code:
https://github.com/GridProtectionAlliance/gsf/commit/ac47ee068532884a552beadd2bf3c2a978e35364

Here is a better naming convention expression now implemented in the openHistorian:

```
{DeviceAcronym}:[?{SignalType.Suffix}=PA[eval{'{PhasorLabel}'.ToUpper().Replace(' ','_')}:ANG]][?{SignalType.Suffix}=PM[eval{'{PhasorLabel}'.ToUpper().Replace(' ','_')}:MAG]][?{SignalType.Source}!=Phasor[{SignalType.Acronym}]][?{SignalIndex}!=-1[{SignalIndex}]]
```

Thanks!
Ritchie

closed 2019-07-15 by ritchiecarroll