GridProtectionAlliance/openPDC #102: Access to STAT using Grafana

logiclabsrl, 2017-12-15:
Hi,
I'm using Grafana to plot data collected by OpenHistorian. With PPA all works fine while I'm not able to access to STAT configuring Grafana Data Source url as http://_ip_address_:6356/api/grafana/ 

openHistorian.exe.config seems to be configured correctly.

Any support will be greatly appreciated.
Thanks
Daniele

ritchiecarroll, 2017-12-15:
These settings work for me:

[image]

Do you happen to be running in debug mode or is this from an installed version?

Thanks,
Ritchie

logiclabsrl, 2017-12-15:
My mistake was to configure HTTP Settings / Access : direct.
With Proxy, it works! 

Thanks
Daniele

closed 2017-12-15 by ritchiecarroll