GridProtectionAlliance/openPDC #103: PPA error

darknba, 2018-01-26:
I have the following errors

"GSF.TimeSeries.TimeSeriesStartupOperations :: PerformTimeSeriesStartupOperations ()" due to exception: Invalid attempt to read when reader is closed.

"Error occurred while logging an error - Object reference not set to an instance of an object"

"[PPA] readerData must contain 43 columns"

I'm running openPDC on linux with mysql.
mysql version: 5.7.21
openPDC 2.3.4
mono 4.6.2

Can someone help me?

StephenCWills, 2018-01-26:
I just tried v2.3 of openPDC on a later version of Mono, and it didn't even build properly. Any chance you could use v2.4 of openPDC? It may fix your issues, and would be easier for us to troubleshoot.

darknba, 2018-01-26:
I already tried with v2.4
the problem occurs when I run openPDC, it connects to the database, but does not save the data in the archive.
The main error I have to solve
"PPA] readerData must contain 43 columns"
"[PPA] Archive write read exception: Sequence contains no elements"

StephenCWills, 2018-01-26:
I wasn't able to reproduce your issue in my dev environment using v2.4 of openPDC. How did you set up your MySQL database? Is there any chance you can use the Configuration Setup Utility on a Windows system to set up a new MySQL database and then simply modify the configuration file on your Linux system to use it? If you do, make sure you match openPDC versions between your Linux and Windows systems.

darknba, 2018-01-30:
I could not make it work in mysql.

I installed Postgres and set up openpdc and it worked.

I'll use postgres

ritchiecarroll, 2018-02-02:
FYI - referenced issue with "[PPA] Archive write read exception: Sequence contains no elements" has been resolved:

https://github.com/GridProtectionAlliance/gsf/commit/9a1ba64e1e7c6a867447c77a29197a031f3eaa35

This was due to metadata file having no initial records.

closed 2018-02-02 by ritchiecarroll