GridProtectionAlliance/openPDC #106: Issue with reading PPA data using Historian Data Viewer tool

xtrememenace, 2018-03-22:
I used the PPA historian (historian output is saved as PPA1) to log the PMU data to the archive folder where openPDC is installed on system A. Based on the .d formatted files stored in the archive folder, I have 165GB total size. I can browse through this data using the historian data viewer.exe tool inside the openPDC folder on system A but if I copy the archive folder and point it to another historian data viewer.exe installed on system B, I can't see any data. I have changed the tool's archive path on system B to point it to the correct location, but I don't see any data. 
[image]
.
.
.
[image]
Do I need to copy any more information to the second openPDC installation on system B to see my archived data from system A or is there any other standalone tool that I can use to view the archived data?

ritchiecarroll, 2018-03-22:
I notice some files are ppa1 and others are ppa?

xtrememenace, 2018-03-22:
Yes. I am using PPA1 as the historian output and it is same as the original historian PPA but with a different name. After installing openPDC on system A, I created a new historian named PPA1 but with same settings as PPA. PPA historian is not enabled. openPDC historian output screenshot from system A:
[image]
Historian Data Viewer on system B (Note, the archive I'm reading in this tool is PPA1 data copied from system A):
[image]

closed 2025-02-08 by ritchiecarroll