GridProtectionAlliance/openPDC #107: Read PMU data from a CSV file using InputAdapter

Dasmunoz123, 2018-05-15:
Hi!

I'm trying to read PMU measurements with an InputAdapter from a CSV file, It is possible?? I create the input adapter but I can't see the measurements asocciated to a PMU device, Do I need create PMU devices in a special way?? 

Thanks

I would appreciate any help

StephenCWills, 2018-05-16:
Create a new device using the `Virtual Device` protocol, then associate the input adapter's measurements with that device.

Dasmunoz123, 2018-05-16:
Here I create a new device (PMU1_PRUEBA) with a virtual device protocol and the csv file in the connection string path;

[image]

After that, I create a mesurement for that new device 

[image]

For last, I create the InputAdapter and try to parameterize, but I don't know if the ColumnMapping field is okay. The thing is, I still can't read the pmu data from my CSV

[image]

Here is the structure of my CSV File, It have a few columns but for the moment I just need to read some of them.

[image]

StephenCWills, 2018-05-19:
Your CSV file needs a header to match your column mappings. For example...

CSV file:
```
Timestamp,PPA:1,PPA:10,PPA:11,PPA:12,PPA:13,PPA:14,PPA:2,PPA:3,PPA:4,PPA:5,PPA:6,PPA:7,PPA:8,PPA:9
635931310170000000,8228,38.47470822,502.5757239,-131.8987716,214.2654052,178.9717246,59.971,0,-0.42,299567.9782,-132.3800155,298562.0961,-132.3510752,230.1433918
635931310170330000,8228,37.93852206,505.0788209,-132.3994386,213.7387727,178.6746272,59.965,0,0.09,299629.9759,-132.7277883,298623.1939,-132.700026,231.7047453
635931310170660000,8228,36.97593536,512.9061342,-133.4377751,212.1360972,178.2193576,59.971,0,-0.72,299813.5209,-133.0427584,298805.9561,-133.0160747,237.4509706
635931310171000000,8228,36.0862927,510.0617702,-134.0836948,212.1936964,177.7745998,59.969,0,1.73,299856.3056,-133.5532424,298848.435,-133.5215979,235.6415516
635931310171330000,8228,35.67089972,513.5070224,-134.566606,211.166653,177.3162248,59.954,0,-0.76,299822.1794,-133.839839,298795.6906,-133.8124328,238.0153716
...
```

CSVInputAdapter connection string:
```
FileName=SHELBY.csv; 
AutoRepeat=True; 
SimulateTimestamp=True; 
TransverseMode=True; 
ColumnMappings={0=Timestamp; 1=PPA:1; 2=PPA:10; 3=PPA:11; 4=PPA:12; 5=PPA:13; 6=PPA:14; 7=PPA:2; 8=PPA:3; 9=PPA:4; 10=PPA:5; 11=PPA:6; 12=PPA:7; 13=PPA:8; 14=PPA:9}
```

ritchiecarroll, 2018-10-17:
Closing this issue - let us know if you still have any trouble...

closed 2018-10-17 by ritchiecarroll

Bayaled, 2020-06-05:
Hi GPA, 

I'm trying to read a CSV to use it as the input of a measurement. I have implemented the steps showed by Stephen and I have not achieve it yet. 

My CSV:
Freq
59.971
59.965
59.971
59.969
59.954
59.971
59.965
59.971
59.969

My Measurement was configured like this:
[image]

My input adapter like this:
[image]

The console present this message:
[image]

and the detail is presented below:
[image]

But the result is my measurement does not present values
[image]

I'll be really grateful for any help from you. 

PD: How to correlate the columns of the CSV with measurements created?

ritchiecarroll, 2020-06-08:
Since you are not using transverse mode, the header row needs to be a little more "fixed", i.e., your CSV file should like the following:
```
Signal ID,Value
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.971
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.965
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.971
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.969
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.954
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.971
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.965
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.971
8cd74c84-39ec-4554-ab54-566971e9d5d8,59.969
```
You can add as many output signal types as you like in this mode - they are just published as they are encountered.

Ritchie