GridProtectionAlliance/openPDC #109: Historian Trending Tool Window Bug

drewvian, 2018-07-18:
I am using the OpenPDC Manager 2.4.120 on Windows 7. When running the Historian Trending Tool and selecting the Open Archive option, the window listing the open archives has a problem. When the list of archives is long, as in too long to view on a single screen, the window grows appropriately to fit all the entries, but there is no scroll bar. That means you can't scroll down to see the lower part of the list, and cannot access the Open or Finish buttons either. Maximizing the window does not help. The only way to actually use the window with the long list is to remove enough entries so that everything can be displayed on a single screen.

drewvian, 2018-07-19:
> What browser and version are you using?

I have to assume you are referring to the OpenPDC version, considering the
issue has nothing to do with a browser. The OpenPDC version is 2.4.120. The
Historian Trending Tool was installed with that version. I have not been
able to locate a separate version number on the Historian Trending Tool.

StephenCWills, 2018-07-19:
I don't think anyone ever anticipated the possibility that any user would ever be opening that many archives at once. So naturally, no scrollbar was ever added to that page. May I ask what the use case is for opening that many archives? Are you sharding your archives or something?

drewvian, 2018-07-19:
I don't really have a good use case. The situation came up from my
misunderstanding of how to use the tool to export data. I was attempting to
export data from a series of archives simultaneously, while keeping the
data discrete. After some use of the tool, I realized that attempting to do
all the archives at once would result in the data being mixed together,
which I don't want.

I have a large series of archives due to some testing where a new historian
was created for each test to assist in separating the data for later
analysis. I would note that by default, when first opening the utility, all
historian archives are present in the list. So if you have a large number
of historians, you cannot even see the Open and Finish buttons.

It might be a good idea to just move the buttons to the top of the window,
that way the buttons are always on the visible part of the window. It would
be a good alternative to adding the scrollbar. Took me awhile to figure out
those buttons even existed, since it was my first use of the application
and wasn't visible until I removed many of the historians from the list.

StephenCWills, 2018-07-19:
I added a MaxHeight to the data grid so that the window won't get too tall. WPF adds the scrollbar to that control automatically. This fix will be available tomorrow in the nightly build version of openPDC (v2.5.30).