GridProtectionAlliance/openPDC #111: Error in openPDC Console and Installation of openPDC

gerarbelizario, 2018-08-01:
Hi, I've installed the openPDC and during the installation an errror poped up saying that there's a problem with the service account. This was when it suggest me to leave in blank if there was automatically wrtitten. I've done what the images says:

[image]
[image]
[image]
[image]

Then, I've used the Wizard and use the files exported from PMU Conenction Tester to configure the openPDC Manager, but It didn't work.

[image]
[image]

ritchiecarroll, 2018-10-17:
The red light at the top right of the openPDC manager shows that the service is not running. Open the Windows "Services" console, scroll down to openPDC and click "start". When openPDC is running, you will not receive service errors in the manager.

gerarbelizario, 2018-10-17:
Hello. I've done it but it says that It has started and inmediately stoped.

[image: image.png]

[image: image.png]

[image: image.png]

ritchiecarroll, 2018-10-17:
You will want to check the "Errorlog.txt" file to see perhaps what happened. Also, if there is nothing relevant there, you can check the Windows event log to see if it reported any errors.

StephenCWills, 2018-10-17:
Given that the `NT SERVICE\openPDC` account could not be added to the `openPDC Admins` group, there's a decent chance that the service account does not have the required permissions to access files in the openPDC installation directory. Lack of permissions is also a known cause for errors that prevent the service from starting.

FYI, I had this issue, among others, when attempting to install openMIC on a system that was configured to be a domain controller. If you are installing on a domain controller, I would recommend creating a domain account to run the service and providing that account with the appropriate group memberships and permissions.

ritchiecarroll, 2018-10-18:
Ah, Stpehen is right - I didn't see your third image - Windows service account will not have rights to installed program files folder. One simple option may be to change Windows service to run as the "Local System account" - see "Log On" tab in openPDC service properties dialog:

[image]

pra12eep, 2019-01-15:
hi...
I have followed the above instructions and it was working fine for few days. but now same problem I facing...openPDC is out of service or stopped from local windows machine. screenshot of error msg is attached and same is written below in case attachment not uploaded.

Windows could not start the openPDC service on Local Computer
Error 14001: The application has failed to start because its side by side configuration in incorrect. please see application event log or use the command line sxstrace.exe tool for more detail.
further if I am trying "sxstrace.exe " in command line it is not opening the command prompt.
Regards
Pradeep

StephenCWills, 2019-01-15:
Hi Pradeep,

It sounds like the openPDC.exe.config file may have been corrupted somehow. The easiest thing for you to do may be to reinstall the openPDC. If you only want to try repairing the one file, you can download the App.config file from the openPDC source code repository, rename the file to openPDC.exe.config, replace your existing openPDC.exe.config (in the openPDC installation directory) with the new one you just downloaded, and then run the Configuration Setup Utility to restore connectivity to your configuration database.

Thanks,
Stephen

pra12eep, 2019-01-21:
Hi Stephen,
Thanks a lot for your quick reply. 
i tried the to repair the earlier installation but still i was getting error like FAILED TO REPAIR.
further, i downloaded the new installation file and trying to install the same but now i am getting below error:

openPDC Setup Wizard ended prematurely because of an error. your system has not been modified. to install this program at a later time, run setup wizard again. click the finish button to exit the setup wizard.

it would be nice if you can write me at pra12eep@gmail.com because i am not able to share the error screenshot here in this platform.
Thank You.

ritchiecarroll, 2019-02-18:
Going to close this issue for now - if this the problem remains, please let us know and we will re-open. 

Thanks,
Ritchie

closed 2019-02-18 by ritchiecarroll