GridProtectionAlliance/openPDC #112: Having trouble installing openPDC on Raspberry Pi 3 Model B+

TTony0014, 2018-10-16:
Hello there,

These days, I have been installing openPDC on Raspberry Pi 3 Model B+ and keep having issues. The instruction I followed is linked from this website: https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/Running_openPDC_on_a_Raspberry_Pi.files/Example-openPDC_on_RaspberryPi-3B_Raspbian_Jessie.pdf

After I installed Mono by following https://www.mono-project.com/download/stable/#download-lin-raspbian, I tried to install openPDC.

By typing "sudo bash install-openPDC.sh -p", in terminal, it shows cannot stat 'openPDC/Build/Output....' (see in attachment). Later, I was noticed to start application, run "sudo mono /opt/openPDC/openPDC.exe -RunAsConsole", and I received "Cannot open assembly '/opt/openPDC/openPDC.exe': No such file or directory"

Did I miss anything while installing Mono or openPDC?

Please help and much appreciated. 

Thank you,
Tony
[image]

StephenCWills, 2018-10-17:
Hello Tony,

Those messages suggest that the openPDC build failed. The build should have produced a log file called `install-openPDC.log` which should give you more detailed messages about why it failed. Feel free to dive into the logs yourself, or you can upload the log file here so I can have a look.

Thanks,
Stephen

TTony0014, 2018-10-17:
Hello Stephen,

Thanks for helping me. I have attached the log file to this post. I am new to Raspberry Pi world so I could not understand what really happen in the log file.

What I find is there is no openPDC.exe file in the targeted direction. Probably as you mentioned, the openPDC did not completely install or build.

Thank you,
Tony
[install-openPDC.log](https://github.com/GridProtectionAlliance/openPDC/files/2488598/install-openPDC.log)

TTony0014, 2018-10-23:
Hello all,

I have kept working on the openPDC installation on Raspberry Pi 3 B+ and 2. Right now, I am trying to install openPDC on Raspberry Pi 2. 

I tried to use the pre-compiled image from https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/Running_openPDC_on_a_Raspberry_Pi.md

But the image does not work. Once I power the Pi 2, the rainbow screen freeze. Green LED shows that the OS is too old. So I have to use other version.

By following issue 96 and 88, I downloaded install-openPDC.zip and try to install with Raspbian Jessie on Raspberry Pi 2.

During installation, still, shows cannot stat some directory.
"Installing openPDC...

cp: cannot stat ‘openPDC/Build/Output/Mono/Applications/openPDC/*’: No such file or directory
mkdir: cannot create directory ‘/opt/openPDC/Database Scripts’: File exists
cp: cannot stat ‘gsf/Build/Output/Mono/Tools/MonoGenCert/MonoGenCert.exe’: No such file or directory
Creating transport layer security keys and certificates...

Cannot open assembly 'MonoGenCert.exe': No such file or directory.
Installing GEP Subscription Tester...

Applying default POSIX configuration to openPDC installation...

openPDC installation complete. To start application, run:
sudo mono /opt/openPDC/openPDC.exe -RunAsConsole"

Then, I tried to run as console. The following issue shows up:
"pi@raspberrypi:~/GPB $ sudo mono /opt/openPDC/openPDC.exe -RunAsConsole
Missing method get_Name in assembly /opt/openPDC/GSF.Core.dll, type System.Security.Claims.ClaimsIdentity
Shell hosted service exception: Object reference not set to an instance of an object"

I saw the same error in issue 96. I wish to know if anyone update the install-openPDC.sh file so far. Mono seems fine now. Maybe installation file includes some error.

Thank you so much.
Tony
[install-openPDC.log](https://github.com/GridProtectionAlliance/openPDC/files/2506681/install-openPDC.log)

StephenCWills, 2018-10-24:
The log reports that ExpressionEvaluator is missing and it can't build GSF.Core. It looks like the build script for GSF isn't restoring NuGet packages. I haven't had the chance to look into it any further than that, but if you can track down the missing dependencies or figure out how to manually restore the NuGet packages, it could get you to the next major hurdle at least.

TTony0014, 2018-10-25:
Good afternoon Stephen and AJ,

I found some info on NuGET package issue and try to attempt to fix it. Meanwhile, I am also thinking about matching the versions of the hardware and software.

Currently, I am installing everything on a Raspberry Pi 2 B and the Raspbian Jessie (4/10/2017 version). I tried using the image provided on the GitHub website, and it will not boot the Pi.

Would you mind sending me an image includes openPDC installed that is working? Or would you please let me know the exact version of RPi and Raspbian I should you. So that I can update my device and software to keep moving.

Thank you very much,
Tony

ajstadlin, 2018-10-28:
FYI:  I just tested the following advanced install-openPDC script in a Raspbian Desktop x86 virtual machine (Hyper-V) and default `-p` option. The result fails with a missing MonoGenCert.exe.

- [Issue #88](https://github.com/GridProtectionAlliance/openPDC/issues/88)
- [Issue #88 proposed advanced install-opendPDC script](https://github.com/GridProtectionAlliance/openPDC/files/681344/install-openPDC.zip)

//AJ

iliesaya, 2019-01-20:
Dear Team, 
any news about this issue ? I am facing the same problem here :
`
>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 5.10.1.20
Copyright (C) 2005-2013 Various Mono authors

Build started 20/01/2019 05:50:57.
__________________________________________________
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Ignoring vcproj 'CSampleAdapters'.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Ignoring vcproj 'TimeSeriesPlatformLibrary'.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Ignoring vcproj 'UnixTools'.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Ignoring vcproj 'SimpleSubscribe'.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Ignoring vcproj 'AdvancedSubscribe'.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Ignoring vcproj 'AverageFrequencyCalculator'.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Ignoring vcproj 'InstanceSubscribe'.
/home/pi/GPA/gsf/Source/Libraries/GSF.InstallerActions/GSF.InstallerActions.csproj:  warning : Could not find project file /usr/lib/mono/xbuil                                     d/Microsoft/WiX/v3.x/Wix.CA.targets, to import. Ignoring.
/home/pi/GPA/gsf/Source/Documentation/GridSolutionsFrameworkHelp/GridSolutionsFrameworkHelp.shfbproj:  warning : Could not find project file /                                     /SandcastleHelpFileBuilder.targets, to import. Ignoring.
/home/pi/GPA/gsf/Source/Documentation/AdapterHelp/AdapterHelp.shfbproj:  warning : Could not find project file //SandcastleHelpFileBuilder.tar                                     gets, to import. Ignoring.
/home/pi/GPA/gsf/Source/Extensions/GPAVSExtension/GPAVSExtension.vbproj:  warning : Could not find project file /usr/lib/mono/xbuild/Microsoft                                     /VisualStudio/v/VSSDK/Microsoft.VsSDK.targets, to import. Ignoring.
/home/pi/GPA/gsf/Source/Libraries/GSF.Core.Shared/GSF.Core.Shared.shproj:  warning : Could not find project file /usr/lib/mono/xbuild/Microsof                                     t/VisualStudio/v/CodeSharing/Microsoft.CodeSharing.Common.Default.props, to import. Ignoring.
/home/pi/GPA/gsf/Source/Libraries/GSF.Core.Shared/GSF.Core.Shared.shproj:  warning : Could not find project file /usr/lib/mono/xbuild/Microsof                                     t/VisualStudio/v/CodeSharing/Microsoft.CodeSharing.Common.props, to import. Ignoring.
/home/pi/GPA/gsf/Source/Libraries/GSF.Core.Shared/GSF.Core.Shared.shproj:  warning : Could not find project file /usr/lib/mono/xbuild/Microsof                                     t/VisualStudio/v/CodeSharing/Microsoft.CodeSharing.CSharp.targets, to import. Ignoring.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Don't know how to handle GlobalSection SharedMSBuildProjectFiles, Ignoring.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Failed to find project bb0caa29-229e-4648-80a0-1474b3a0facf
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Failed to find project 2a542be8-8d17-44c3-bcd2-768df480ff82
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Failed to find project 63af9800-2cde-41f7-9706-76ef0ae181ce
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Failed to find project bb15820e-b4b7-4dcb-887e-b78aa648fe3d
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Failed to find project 6dca52be-054a-4f25-b08a-3195543a169c
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Failed to find project 3ab7fb78-65fe-4526-991f-532f39d9a26a
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Failed to find project 87f19b27-127e-4241-a442-345ac1f71fe0
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Don't know how to handle GlobalSection ExtensibilityGlobals, Ignoring.
/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln:  warning : Don't know how to handle GlobalSection TestCaseManagementSettings, Ignoring.
Project "/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln" (default target(s)):
        Target ValidateSolutionConfiguration:
                Building solution configuration "Mono|Any CPU".
        Target Build:
                Project "/home/pi/GPA/gsf/Source/Libraries/GSF.Core/GSF.Core.csproj" (default target(s)):
                        Target PrepareForBuild:
                                Configuration: Mono Platform: AnyCPU
                        Target ResolveAssemblyReferences:
/usr/lib/mono/xbuild/14.0/bin/Microsoft.Common.targets:  warning : Reference 'Antlr3.Runtime, Version=3.5.0.2, Culture=neutral, PublicKeyToken                                     =eb42632606e9261f, processorArchitecture=MSIL' not resolved
                                For searchpath {CandidateAssemblyFiles}
                                Warning: {CandidateAssemblyFiles} not supported currently
                                For searchpath {HintPathFromItem}
                                Considered ../../Dependencies/NuGet/Antlr.3.5.0.2/lib/Antlr3.Runtime.dll, but it does not exist.
                                For searchpath {TargetFrameworkDirectory}
                                Considered target framework dir /usr/lib/mono/4.6-api/, assembly named 'Antlr3.Runtime' not found.
[... cut ...]
                                Copying file from '/home/pi/GPA/gsf/Source/Libraries/GSF.ASN1/obj/Mono/GSF.ASN1.dll' to '/home/pi/GPA/gsf/Buil                                     d/Output/Mono/Libraries/GSF.ASN1.dll'
                Done building project "/home/pi/GPA/gsf/Source/Libraries/GSF.ASN1/GSF.ASN1.csproj".
        Task "MSBuild" execution -- FAILED
        Done building target "Build" in project "/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln".-- FAILED
Done building project "/home/pi/GPA/gsf/Source/GridSolutionsFramework.sln".-- FAILED

Build FAILED.
`

ajstadlin, 2019-08-29:
Thank you for the follow-up and helpful troubleshooting guidance.

I am prepping to try the new Raspberry Pi 4 with 4GB RAM, hoping if the openPDC build does not work in the Raspbian OS, there may now be a supported Ubuntu Server 64bit OS to run on it.  I will post back with the results.

ritchiecarroll, 2019-08-29:
Just a note - @StephenCWills  has a found a few minor build issues and is working on fix! :-)

lhartmann, 2020-04-03:
Same issue, on docker containers with ubuntu or debian. The POSIX installer script seems to be outdated and broken.

README.md, in the "Deployment" section for POSIX refers to [this document](https://gpags.sharepoint.com/TeamSite/_layouts/15/WopiFrame.aspx?guestaccesstoken=ADyQzHPxsfTh9qs4glPelL78SoBA1pTJV1%2fWy6b0ct4%3d&docid=08819043371f24a089e4924e86525dd69&action=view), which refers to [this script](https://www.gridprotectionalliance.org/Products/openPDC/Scripts/install-openPDC.sh), which tries to install openPDC version 2.5 (current stable is 2.7.4).

ajstadlin, 2020-04-05:
There is an alternate script with some additional advanced options for getting other openPDC versions.  Submitted by [avs009](https://github.com/avs009) - [Issue #88 proposed advanced install-openPDC script](https://github.com/GridProtectionAlliance/openPDC/files/681344/install-openPDC.zip)  The actual "stable version" is a constant in the scripts and is easy to change.

I am interested in running openPDC on the Raspberry Pi platform - especially with the release of RPi-4 with plenty of RAM and 64bit Linux versions in the works.  However, I am personally unable to work on it at this time.

//AJ

Milozpp, 2020-11-26:
Hi to all,
i'm trying to install openPDC in my vm running ubuntu20.04 but i'am facing a problem(one is exatcly the same posted in the post) namely 'cannot open assembly opt/openPDC/openPDC.exe: no such file or directory'. I saw the log file generated after the bash command sudo bash install-openPDC.sh has been throw and it reports a lot of entries that i found in old posts. My questions are: it is right try  to install openPDC in my ubuntu20.04? It is a compatible platform?  Could someone explains me what it the right procedure/configuration to fix my problems? TY

ashu704, 2021-03-06:
> Thank you for the follow-up and helpful troubleshooting guidance.
> 
> I am prepping to try the new Raspberry Pi 4 with 4GB RAM, hoping if the openPDC build does not work in the Raspbian OS, there may now be a supported Ubuntu Server 64bit OS to run on it. I will post back with the results.

Tried installing on pi4 it shows the same error as this post

valentin-velev, 2021-11-07:
The problem "cp: cannot stat 'MonoGenCert.exe': No such file or directory" appears because 'MonoGenCert.exe' is missing. Missing 'MonoGenCert.exe' file causes the following line in the script to fail: 
cp -v MonoGenCert.exe ${destDir}/ >> ${log} 
'MonoGenCert.exe' can be compiled from GSF sources and then placed in the folder of 'install-openPDC.sh'. I have compiled 'MonoGenCert.exe' with Visual Studio in Windows.

At the end of the script 'install-openPDC.sh' I've got problem with the removing of 'gsf' folder in the case when 'gsf' is not downloaded and 'gsf' folder is not created. I have added a check to avoid failing of the remove action:
if [ $downloadcompileGSF -ne 0 ]; then                                                                    
    rm -rv gsf >> ${log}                                                                              
fi

ritchiecarroll, 2022-04-01:
Try the latest release:
https://github.com/GridProtectionAlliance/openPDC/releases/tag/v2.9.148

closed 2022-04-01 by ritchiecarroll