GridProtectionAlliance/openPDC #114: Some Wiki Documentation URL not access

avs009, 2018-11-20:
Hi, Some links in [Wiki Documentation](https://github.com/GridProtectionAlliance/openPDC/wiki/Documentation) not open to read, and say "You do not have permission to update this wiki." or not exist no matter if you are logged or not.

The following links say "You do not have permission to update this wiki.":

1. [Getting Started](https://github.com/GridProtectionAlliance/openPDC/wiki/GettingStarted)
2. [Frequently Asked Questions](https://github.com/GridProtectionAlliance/openPDC/wiki/FAQ)
3. [Use and Configuration Guides](https://github.com/GridProtectionAlliance/openPDC/wiki/UseAndConfigurationGuides)
4. [Overview of the openPDC Management System](https://github.com/GridProtectionAlliance/openPDC/wiki/IntroducingTheOpenPDCManager)
5. [How to Use the openPDC Manager](https://github.com/GridProtectionAlliance/openPDC/wiki/OpenPDCManagerConfiguration)
6. [Running openPDC on Linux / Mac](https://github.com/GridProtectionAlliance/openPDC/wiki/RunningOpenPDCOnLinuxAndMac)
7. [Running openPDC on a Raspberry Pi](https://github.com/GridProtectionAlliance/openPDC/wiki/RunningOpenPDCOnARaspberryPi)
8. [Data Quality Reporting Services](https://github.com/GridProtectionAlliance/openPDC/wiki/OpenPDCDataQualityReports)
9. [Custom Point Tag Naming Convention](https://github.com/GridProtectionAlliance/openPDC/wiki/CustomPointTagNamingConvention)
10. [Enabling Security for Historian Web Services](https://github.com/GridProtectionAlliance/openPDC/wiki/EnablingSecurityForHistorianWebServices.md)
11. [The GEP Subscription Tester](https://github.com/GridProtectionAlliance/openPDC/wiki/GEPSubscriptionTester.md)
12. [Running the openPDC in Virtual Machine](https://github.com/GridProtectionAlliance/openPDC/wiki/RunningOpenPDCInVirtualMachine)
13. [Moving Historian to Another Location](https://github.com/GridProtectionAlliance/openPDC/wiki/MoveLocalHistorianToAnotherFolder)
14. [Using a "Gateway Style Connection" between openPDCs and/or openPGs](https://github.com/GridProtectionAlliance/openPDC/wiki/UsingAGatewayStyleConnectionBetweenOpenPDCsAndForOpenPGs)
15. [Controlling UDP data loss](https://github.com/GridProtectionAlliance/openPDC/wiki/ControllingUDPDataLoss)
16. [Remote Console Security](https://github.com/GridProtectionAlliance/openPDC/wiki/RemoteConsoleSecurity)
17. [Help Me Choose Diagrams](https://github.com/GridProtectionAlliance/openPDC/wiki/HelpMeChooseDiagrams)
18. [Configuration File Settings](https://github.com/GridProtectionAlliance/openPDC/wiki/ConfigFile)
19. [Setting for Uniform CPU Utilization](https://github.com/GridProtectionAlliance/openPDC/wiki/SettingsForUniformCPUUtilization)
20. [openPDC Console Commands to Adjust Configuration Settings](https://github.com/GridProtectionAlliance/openPDC/wiki/ConfigurationCommands)
21. [openPDC OSI-PI Adapters](https://github.com/GridProtectionAlliance/openPDC/wiki/OSI-PIAdapters)
22. [Data Quality Monitoring Adapters](https://github.com/GridProtectionAlliance/openPDC/wiki/DataQualityMonitoring)
23. [Adapter Connection String Syntax](https://github.com/GridProtectionAlliance/openPDC/wiki/ConnectionStrings)
24. [Distributed Historian Setup Notes](https://github.com/GridProtectionAlliance/openPDC/wiki/HistorianDistributionNotes)
25. [Automated Archive Data Recovery Operation](https://github.com/GridProtectionAlliance/openPDC/wiki/AutomatedArchiveDataRecoveryOperation)
26. [Automated Connection Failover Operation](https://github.com/GridProtectionAlliance/openPDC/wiki/AutomatedConnectionFailover.md)
27. [Adjusting Output Stream Labels to Meet ISO Naming Standard](https://github.com/GridProtectionAlliance/openPDC/wiki/AdjustingOutputStreamLabelsToMeetISONamingConvention)
28. [How to Manually Configure the openPDC](https://github.com/GridProtectionAlliance/openPDC/wiki/ManualConfiguration)
29. [How to Bulk Apply line-to-line Sqrt(3) Adjustment to all Voltage Magnitudes](https://github.com/GridProtectionAlliance/openPDC/wiki/HowToBulkApplyLineToLineSqrt3AdjustmentToAllVoltageMagnitudes)
30. [Single Page Overview](https://github.com/GridProtectionAlliance/openPDC/wiki/openPDCOverview)

  Developer's Documentation

31. [Getting Started](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersGettingStarted)
32. [Frequently Asked Questions](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersFrequentlyAskedQuestions)
33. [Build the openPDC Manager](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersBuildTheOpenPDCManager)
34. [Data Access Options for the openPDC](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersDataAccessOptions)
35. [Device to Data in 5 Easy Steps](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersDeviceToDataIn5EasySteps)
36. [High-level Code Structure and Class Relationships](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersAboutTheCode)
37. [How to Create a Custom Adapter](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersCustomAdapters)
38. [Two Custom Adapter Examples](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersTwoCustomAdapterExamples)
39. [Multiple Adapter Synchronization](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersMultipleAdapterSynchronization)
40. [Automated Phasor Tag Naming Convention](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersAutomatedPhasorTagNamingConvention)
41. [About the Signal Reference Field](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersAboutTheSignalReference)
42. [Processing openPDC data with Hadoop](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersUsingHadoop)
43. [Code Change Notes](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersCodeChangeNotes)

  Bottom notes:

44. [Nightly Builds](https://github.com/GridProtectionAlliance/openPDC/wiki/NightlyBuilds)
45. [ integrated within Visual Studio](https://github.com/GridProtectionAlliance/openPDC/wiki/DevelopersGettingStarted)

The following links say "Not Found (Error 404):
1. [Default openPDC Ports](https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/FAQ.files/Default_openPDC_Ports.rtf)

The following links is old site:
1. [PMU Connection Tester](http://pmuconnectiontester.codeplex.com/documentation)

Related Issues:
#19 , #28 .

ritchiecarroll, 2018-11-20:
I'm not getting this error - even if I open in an incognito window?

ritchiecarroll, 2018-11-20:
I sent you an invitation - let me know what you find... Thanks!

avs009, 2018-11-20:
Just for take sure what i will do, these pages exist in folder [openPDC/tree/master/Source/Documentation/wiki](https://github.com/GridProtectionAlliance/openPDC/tree/master/Source/Documentation/wiki).

Can i just update the link for that, or is best copy and paste the content?

In the second option, every update need to be done inside git wiki by me or another person.

ritchiecarroll, 2018-11-20:
Try updating the link and see if that works for you... but let's just try one or two since I am not sure what's broken here - you might fix it for you and break it for me :-)

AJ did all this work in converting this documentation over from CodePlex - so he's a good contact - and I am sure he'd appreciate some help on maintaining this work (just as we do!)

Thanks,
Ritchie

ritchiecarroll, 2018-11-20:
Looks like your updates are working - thanks! I'll let you keep at it and I will close this issue for now.

Ritchie

closed 2018-11-20 by ritchiecarroll

avs009, 2018-11-20:
Now i see more clarification about what we can do with the git wiki.
They already has a private repository in [Git Wiki Repository](https://github.com/GridProtectionAlliance/openPDC.wiki.git) and the best way to do some is migrate from master [openPDC/tree/master/Source/Documentation/wiki](https://github.com/GridProtectionAlliance/openPDC/tree/master/Source/Documentation/wiki).

ritchiecarroll, 2018-11-20:
That would work - but there would be "many" links to update and documentation to copy - but if you are up for it? @StephenCWills @ajstadlin  thoughts?

avs009, 2018-11-20:
I will do some tests with a private repository in wiki repo first, to take sure if need to update links or not.

StephenCWills, 2018-11-20:
I believe that the primary limitation with the wiki repository is that you can't easily do forks, and you can't submit pull requests. The best workaround I've seen involves creating another separate repo to maintain the documentation and then syncing that repo with the wiki repo, and that sounds like a pain. Also, it appears there is an option to expose the wiki repo to any and all GitHub users, which I guess allows for more collaboration at the expense of security. That said, since we tend to give out write access to our repositories, this may not be as big of an issue as having to update all the links.

We also have the option of simply replacing the homepage of the wiki repository with a link to the appropriate location in the main repo. This would avoid the pull request problem, the effort of updating links, and the issue of maintaining the documentation in two separate places. The only real drawback would be that the history for the documentation will be kept in the main repo instead of being cleanly separated in its own repo, but Git tools tend to be pretty good about allowing you to filter those down.

As a third option, we could turn off the wiki repo for this repository and rely on the fact that the README.md has a link to the appropriate location for the documentation in the main repo. It's not front-and-center like the Wiki tab is, but it is the first place people tend to look for information when they browse to a new project. This has the same pros and cons as the previous option.

I think it would be okay with opening up the wiki to all GitHub users just to make it as easy as possible to contribute. We'd just have to be sure to maintain a local copy just in case.

avs009, 2018-11-20:
I am back, the migration is possible, but all the links must be updated without extension ".md", a simple "sed command" can do this.

As @StephenCWills says, the wiki repo not able to submit merge request, users always see the master version of repo.

Is able to create branches, but need to manual merge with master, there's no web interface to see changes between branches.

All users are able to clone the wiki repo, i think if they need update some terms, simple download the page and submit the changed file or a patch in an issue.

I think the best option is migrate all documentation to wiki repo, that's is worst option because need a large review but I'm willing to do.

StephenCWills, 2018-11-20:
@avs009, If you're willing, I say go for it. I like the idea of having the documentation in the wiki repo.