GridProtectionAlliance/openPDC #115: not able to receive data from PMU

pra12eep, 2018-12-26:
Dear All, I have setup the openPDC system in the lab and I was able to get the output stream of 2 PMU in openPDC after checking them through PMU Connection Tester. 
After few days I was trying to connect the 3rd PMU and system showed a message that "Application is currently disconnected from service" and I am not able to get the out of any PMU in openPDC. 
 I have screenshots of the error msgs which i trying to upload but system is giving error that we can't process that file.

Kindly guide me.

StephenCWills, 2019-01-02:
The openPDC Manager typically displays that message when the service is offline. I recommend checking the Windows Services Manager to determine the current state of the openPDC service.

https://www.thewindowsclub.com/open-windows-services

pra12eep, 2019-01-08:
its working now.
Thank you Stephen!

closed 2019-01-08 by pra12eep