GridProtectionAlliance/openPDC #116: Identification Error

iradevic, 2019-01-09:
Hello,

I am facing the identification issue when running openPDCManager. Please, find the screenshot of the issue attached. I would kindly ask for your opinion about the possible solution. 

Thank you

Isidora

[image]

StephenCWills, 2019-01-09:
Hello Isidora,

I just have a few questions that may help to pin down the resolution.

1. Is this a clean install of the openPDC, or has it been upgraded?
2. Does the NodeID field in the `openPDCManager.exe.config` file match the NodeID field in `openPDC.exe.config`?
3. Are you using MySQL for your configuration database?

Thanks,
Stephen

iradevic, 2019-01-09:
Hello Stephen,

Thank you for your prompt reply.
These are answers:
1. This is a clean install of the openPDC
2. NodeID fields match
3. Yes, I am using MySQL

I hope the answers filter some solution! 

Thanks,

Isidora

StephenCWills, 2019-01-09:
This issue has been reported to occur with certain versions of the MySQL Connector/NET driver. The 6.9.12 version of MySQL Connector/NET (available by clicking the "Looking for previous GA releases?" link) is the one that works best with our software.

iradevic, 2019-01-10:
Thank you, problem solved in the way you suggested!

closed 2019-01-10 by StephenCWills