GridProtectionAlliance/openPDC #117: How to use openPDC Data for further studies.

pra12eep, 2019-02-18:
Can we view live PMU data connected to the openPDC in tabular form on a single platform? Right now we are using CSV export feature and getting the data in the form of CSV but the data is not live. For example if we are fetching 5 min data, the data being reflected in the CSV is the past 5 min data starting from now. Even while using the Historian Trending Tool, the live data is not getting reflected. Kindly suggest a method to view live data on a single platform.

ritchiecarroll, 2019-03-01:
Why not use Grafana - it supports tables and you can refresh on an interval?

pra12eep, 2019-03-11:
Hello Ritchie,
Our requirement is to get real time data (with delay as less as 100ms) in
the form of numbers not graph from the PMUs which can be stored in a local
database. We want to then process the stored data (with delay again in tens
of milliseconds) for further analysis.

Regards,
Pradeep

-- 

Thank You

ritchiecarroll, 2019-03-11:
I think openHisorian would be ideal for this purpose

closed 2022-04-01 by ritchiecarroll