GridProtectionAlliance/openPDC #119: Output stream issue

pietropau, 2019-03-01:
The issue occurs when you configure an output stream when the input PMU has a large number of phasors. In the example we have 24 phasor, look at phasor PM9 ( Voltage positive sequence 405 KV).

[image]

I configure output stream with only positive sequence.

[image]

As you can see in the output stream the voltage PM9 is 235Kv, it seems that It's aiming to wrong phasor ( phase B)

[image]

Thanks
Pietro

StephenCWills, 2019-03-01:
Hi Pietro,

Are you saying that you manually deleted the phasors in the Manage Phasors page of openPDC Manager?

Thanks,
Stephen

pietropau, 2019-03-01:
Yes

Thanks
Pietro

ritchiecarroll, 2019-03-01:
Hi Pietro - we are assuming this is unrelated to the other issue with a read-only domain controller:

https://discussions.gridprotectionalliance.org/t/openhistorian-web-interface/383/12

Also, Giorgio mentioned in an e-mail you were going to post some logs related to read-only domain controller...

Thanks!
Ritchie