GridProtectionAlliance/openPDC #120: Connecting 1 PMU to multiple PDCs

asthaIITD, 2019-04-02:
I am trying to connect 1 PMU to 2 PDCs(openPDC) running on different systems in same network. 

Observation
PMU doesn't remain continuously connected to either PDC. But when I stop openPDC service on 1 system, PMU remains continuously connected to other system.

Request you all to help me with its explanation and solution regarding how can I make PMU connection continuous to both PDC

ritchiecarroll, 2019-04-02:
What kind of PMU are you using? Some PMUs only like, or can be configured for, a single connection.

asthaIITD, 2019-04-02:
@ritchiecarroll Sir, I am using Real-Time Digital Simulator(RTDS) GTNET Software PMUs.

ritchiecarroll, 2019-04-02:
The simulator may not like multiple connections - I would suggest using the stream splitter:
https://github.com/GridProtectionAlliance/phasorsplitter

After installing stream splitter, connect stream splitter to RTDS stream - then connect PDCs to stream splitter.

asthaIITD, 2019-04-03:
Thankyou So much @ritchiecarroll Sir.

asthaIITD, 2019-04-08:
@ritchiecarroll Sir, It would be great if you can share some documentation related to configuring Stream splitter. I could not find it.

ritchiecarroll, 2019-08-29:
A little here:  https://archive.codeplex.com/?p=phasorsplitter
Ask questions here: https://discussions.gridprotectionalliance.org/c/gpa-products/stream-splitter

closed 2019-08-29 by ritchiecarroll