GridProtectionAlliance/openPDC #121: [Kafka adapter] Failed to serialize metadata records: cast is not valid

tzonghao, 2019-05-14:
(Version 2.6 or 2.7) When Kafka adapter output is enabled, the error is reported and no metadata is produced in Kafka topic:

`[KAFKA] Failed to serialize current time-series metadata records: Specified cast is not valid.`

ritchiecarroll, 2019-05-15:
I don't think we've tested the Kafka adapter in a while - my guess is that we have changed-up measurement metadata functionality since this adapter was written and there's a small bug here now.

ritchiecarroll, 2019-05-15:
FYI - I looked at the code where this error occurred and it's not obvious to me why this is failing. Perhaps it's related to the type of database  you are using. What is your openPDC configuration database type??

tzonghao, 2019-05-15:
I'm using SQLite.

ritchiecarroll, 2019-05-15:
I can think of two things (1) CHAR in SQLite is not properly converting to string, or (2) DateTime is having cultural issues.

Is it possible to try another database to see if the error persists, such as, SQL Server Express?

Thanks,
Ritchie

tzonghao, 2019-05-15:
I can try PostgreSQL and maybe some other databases, sometime tomorrow. 
Thanks for looking into this.

tzonghao, 2019-05-15:
I tried to use both PostgreSQL and MySQL, but openPDC failed to connect/setup with either, so there's no way to tell if they'd have the same issue. However, SQL Server Express 2017 works out of the box and does not have the same issue as with SQLite.