GridProtectionAlliance/openPDC #124: Analog-signal-only virtual device has 8000 status flags

DumbGrid, 2019-07-15:
Hey guys, I have a question about the virtual device.

Background:  We have an adapter from Quanta Technology (the RVII, voltage stability adapter.)  It takes some inputs, does some calculations and outputs some analog signals. We want to send these analog signals to downstream applications – PhasorPoint for display.

Before:            I’ve put the these analog signals back into the same PMU that was used as input, and it worked.

Now:               We want to separate them from the originating PMU for easy maintenance. So I created a virtual device to host these analog signals. 

Problem:        My problem now is the analog signals went through, but the Status Flag is 8000 (data invalid.) When I checked the device measurements, there were no Status Flag, so I’m guessing it’s automatically added by the C37 protocol (same as frequency and ROCOF?) and it’s default value is 8000? I tried manually adding a Status Flag measurement but since there is no source signal feeding it, no difference is made. 

Any suggestions? 

Thanks!

ritchiecarroll, 2019-07-15:
You will need to provide a value to the measurement - my suggestion is to just use the Dynamic Calculator adapter and assign it a constant value.

Here are a few notes:

https://github.com/GridProtectionAlliance/openPDC/blob/master/Source/Documentation/wiki/Use_and_Configuration_Guides.files/Dynamic_Calculations.pdf

https://discussions.gridprotectionalliance.org/t/how-to-use-adapter-dynamic-calculator/401

Also see the following documentation on how to write expressions:

https://github.com/mparlak/Flee/wiki [github.com]

https://www.codeproject.com/Articles/19768/Flee-Fast-Lightweight-Expression-Evaluator [codeproject.com]

Thanks!
Ritchie

DumbGrid, 2019-07-15:
Thank you Ritchie.

closed 2019-08-29 by ritchiecarroll