GridProtectionAlliance/openPDC #126: openPDC on linux has issues bringing up web server

zobo, 2019-11-25:
When starting openPDC with mono (6.4.0.198) on linux I get a bunch of WPF errors like this:
```Failed to initialize web hosting: Could not load type of field 'System.Windows.VisualStateChangedEventArgs:_control' (2) due to: Could not load file or assembly 'PresentationFramework, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies.

Failed to initialize web hosting: Could not set up parent class, due to: Invalid type Microsoft.Windows.Controls.DataGrid for instance field Microsoft.Windows.Controls.DataGridColumn:_dataGridOwner assembly:/root/openpdc-new/out/WPFToolkit.dll type:DataGridColumn member:(null)

Failed to initialize web hosting: Could not set up parent class, due to: Invalid type Microsoft.Windows.Controls.DataGridColumn for instance field Microsoft.Windows.Controls.DataGridColumnEventArgs:_column assembly:/root/openpdc-new/out/WPFToolkit.dll type:DataGridColumnEventArgs member:(null)

Failed to initialize web hosting: Invalid type Microsoft.Windows.Controls.Calendar for instance field Microsoft.Windows.Controls.CalendarBlackoutDatesCollection:_owner

Failed to initialize web hosting: Could not set up parent class, due to: VTable setup of type Microsoft.Windows.Controls.ControlBehavior failed assembly:/root/openpdc-new/out/WPFToolkit.dll type:ControlBehavior member:(null)

Failed to initialize web hosting: Invalid type Microsoft.Windows.Controls.DataGridColumn for instance field Microsoft.Windows.Controls.DataGridCellEditEndingEventArgs:_dataGridColumn

Failed to initialize web hosting: Invalid type Microsoft.Windows.Controls.DataGridColumn for instance field Microsoft.Windows.Controls.DataGridAutoGeneratingColumnEventArgs:_column

Failed to initialize web hosting: Could not load type of field 'Microsoft.Windows.Controls.DataGridRowDetailsEventArgs:<DetailsElement>k__BackingField' (0) due to: Could not load file or assembly 'PresentationFramework, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies.

Failed to initialize web hosting: Invalid type Microsoft.Windows.Controls.DataGridColumn for instance field Microsoft.Windows.Controls.DataGridPreparingCellForEditEventArgs:_dataGridColumn

Failed to initialize web hosting: Could not set up parent class, due to: VTable setup of type Microsoft.Windows.Controls.ControlBehavior failed assembly:/root/openpdc-new/out/WPFToolkit.dll type:ControlBehavior member:(null)

Failed to initialize web hosting: Invalid type Microsoft.Windows.Controls.DataGridRow for instance field Microsoft.Windows.Controls.DataGridRowEditEndingEventArgs:_dataGridRow

Failed to initialize web hosting: Invalid type Microsoft.Windows.Controls.Calendar for instance field Microsoft.Windows.Controls.SelectedDatesCollection:_owner

Failed to initialize web hosting: Could not set up parent class, due to: Could not set up parent class, due to: Invalid type Microsoft.Windows.Controls.DataGrid for instance field Microsoft.Windows.Controls.DataGridColumn:_dataGridOwner assembly:/root/openpdc-new/out/WPFToolkit.dll type:DataGridColumn member:(null) assembly:/root/openpdc-new/out/WPFToolkit.dll type:DataGridBoundColumn member:(null)

Failed to initialize web hosting: Could not load type of field 'System.Windows.VisualStateGroup:_currentStoryboards' (2) due to: Could not load file or assembly 'PresentationFramework, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies.
```

However after roughly 4 minutes the errors stop and the web server comes up and works. I can login and use the interface...

I tried to pinpoint where the exceptions come from but could't so far.. Here are some stack taces:
```(Outer Exception)
Date and Time:         11/25/2019 11:53:46 AM
Machine Name:          comsensus
Machine IP:            10.0.2.15
Machine OS:            Unix 4.9.0.7

Application Domain:    openPDC.exe
Assembly Codebase:     root/openpdc-new/out/openPDC.exe
Assembly Full Name:    openPDC, Version=2.8.40.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.8.40.0
Assembly Build Date:   11/18/2019 1:59:21 PM
.Net Runtime Version:  4.0.30319.42000

Exception Source:
Exception Type:        System.InvalidOperationException
Exception Message:     Failed to initialize web hosting: Invalid type Microsoft.Windows.Controls.DataGridColumn for instance field Microsoft.Windows.Controls.DataGridCellEditEndingEventArgs:_dataGridColumn

---- Stack Trace ----

[11/25/2019 11:53:48 AM] (Inner Exception)
Date and Time:         11/25/2019 11:53:48 AM
Machine Name:          comsensus
Machine IP:            10.0.2.15
Machine OS:            Unix 4.9.0.7

Application Domain:    openPDC.exe
Assembly Codebase:     root/openpdc-new/out/openPDC.exe
Assembly Full Name:    openPDC, Version=2.8.40.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.8.40.0
Assembly Build Date:   11/18/2019 1:59:21 PM
.Net Runtime Version:  4.0.30319.42000

Exception Source:
Exception Type:        System.TypeLoadException
Exception Message:     Invalid type Microsoft.Windows.Controls.DataGridColumn for instance field Microsoft.Windows.Controls.DataGridAutoGeneratingColumnEventArgs:_column

---- Stack Trace ----
[... 306 lines cut ...]
       openPDC.exe: N 00307
   System.Web.Http.HttpConfiguration.EnsureInitialized()
       openPDC.exe: N 00067
   openPDC.Startup.Configuration(app As IAppBuilder)
       Startup.cs: Ln 0163, Col 13, IL 0766
   Missing Method   System.Reflection.RuntimeMethodInfo.Invoke(obj As Object, invokeAttr As BindingFlags, binder As Binder, parameters As Object[], culture As CultureInfo)
       openPDC.exe: N 00276

(Outer Exception)
Date and Time:         11/25/2019 11:53:36 AM
Machine Name:          comsensus
Machine IP:            10.0.2.15
Machine OS:            Unix 4.9.0.7

Application Domain:    openPDC.exe
Assembly Codebase:     root/openpdc-new/out/openPDC.exe
Assembly Full Name:    openPDC, Version=2.8.40.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.8.40.0
Assembly Build Date:   11/18/2019 1:59:21 PM
.Net Runtime Version:  4.0.30319.42000

Exception Source:
Exception Type:        System.InvalidOperationException
Exception Message:     Failed to initialize web hosting: Could not load type of field 'System.Windows.VisualStateManager:DurationZero' (2) due to: Could not load file or assembly 'PresentationCore, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies.
```

zobo, 2019-11-25:
Similarly this creeps up:
```[11/25/2019 12:19:05 PM] Failed to initialize web hosting: Could not load type of field 'OSIsoft.AF.PI.PIServer:_channel' (16) due to: Could not load file or assembly 'OSIsoft.PI.Net.Core, Version=1.5.0.0, Culture=neutral, PublicKeyToken=c3309f0734ba2805' or one of its dependencies.
```
```
[11/25/2019 12:19:05 PM] (Inner Exception)
Date and Time:         11/25/2019 12:19:05 PM
Machine Name:          comsensus
Machine IP:            10.0.2.15
Machine OS:            Unix 4.9.0.7

Application Domain:    openPDC.exe
Assembly Codebase:     root/openpdc-new/out/openPDC.exe
Assembly Full Name:    openPDC, Version=2.8.40.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.8.40.0
Assembly Build Date:   11/18/2019 1:59:21 PM
.Net Runtime Version:  4.0.30319.42000

Exception Source:
Exception Type:        System.TypeLoadException
Exception Message:     Could not load type of field 'OSIsoft.AF.PI.PIServer:_channel' (16) due to: Could not load file or assembly 'OSIsoft.PI.Net.Core, Version=1.5.0.0, Culture=neutral, PublicKeyToken=c3309f0734ba2805' or one of its dependencies.

---- Stack Trace ----
   Missing Method   System.RuntimeType.IsAssignableFrom(c As Type)
       openPDC.exe: N 00195
   System.Web.Http.Dispatcher.DefaultHttpControllerTypeResolver.IsControllerType(t As Type)
       openPDC.exe: N 00198
   System.Web.Http.Dispatcher.DefaultHttpControllerTypeResolver.<GetControllerTypes>b__7_0(x As Type)
       openPDC.exe: N 00086
   System.Linq.WhereArrayIterator`1.MoveNext()
       openPDC.exe: N 00179
   System.Collections.Generic.List`1.AddEnumerable(enumerable As IEnumerable`1)
       openPDC.exe: N 00171
   System.Collections.Generic.List`1.InsertRange(index As Int32, collection As IEnumerable`1)
       openPDC.exe: N 00570
   System.Collections.Generic.List`1.AddRange(collection As IEnumerable`1)
       openPDC.exe: N 00062
   System.Web.Http.Dispatcher.DefaultHttpControllerTypeResolver.GetControllerTypes(assembliesResolver As IAssembliesResolver)
       openPDC.exe: N 00858
   System.Web.Http.Dispatcher.HttpControllerTypeCache.InitializeCache()
       openPDC.exe: N 00111
   System.Lazy`1.ViaFactory(mode As LazyThreadSafetyMode)
       openPDC.exe: N 00358
   System.Lazy`1.ExecutionAndPublication(executionAndPublication As LazyHelper, useDefaultConstructor As Boolean)
       openPDC.exe: N 00151
   System.Lazy`1.CreateValue()
       openPDC.exe: N 00164
   System.Lazy`1.get_Value()
       openPDC.exe: N 00062
   System.Web.Http.Dispatcher.HttpControllerTypeCache.get_Cache()
       openPDC.exe: N 00058
   System.Web.Http.Dispatcher.DefaultHttpControllerSelector.InitializeControllerInfoCache()
       openPDC.exe: N 00315
   System.Lazy`1.ViaFactory(mode As LazyThreadSafetyMode)
       openPDC.exe: N 00358
   System.Lazy`1.ExecutionAndPublication(executionAndPublication As LazyHelper, useDefaultConstructor As Boolean)
       openPDC.exe: N 00151
   System.Lazy`1.CreateValue()
       openPDC.exe: N 00164
   System.Lazy`1.get_Value()
       openPDC.exe: N 00062
   System.Web.Http.Dispatcher.DefaultHttpControllerSelector.GetControllerMapping()
       openPDC.exe: N 00066
   System.Web.Http.Routing.AttributeRoutingMapper.AddRouteEntries(collector As SubRouteCollection, configuration As HttpConfiguration, constraintResolver As IInlineConstraintResolver, directRouteProvider As IDirectRouteProvider)
       openPDC.exe: N 00138
   System.Web.Http.Routing.<>c__DisplayClass1_1.<MapAttributeRoutes>b__1()
       openPDC.exe: N 00159
   System.Web.Http.Routing.RouteCollectionRoute.EnsureInitialized(initializer As Func`1)
       openPDC.exe: N 00095
   System.Web.Http.Routing.<>c__DisplayClass1_0.<MapAttributeRoutes>b__0(config As HttpConfiguration)
       openPDC.exe: N 00307
   System.Web.Http.HttpConfiguration.EnsureInitialized()
       openPDC.exe: N 00067
   openPDC.Startup.Configuration(app As IAppBuilder)
       Startup.cs: Ln 0163, Col 13, IL 0766
   Missing Method   System.Reflection.RuntimeMethodInfo.Invoke(obj As Object, invokeAttr As BindingFlags, binder As Binder, parameters As Object[], culture As CultureInfo)
       openPDC.exe: N 00276

(Outer Exception)
Date and Time:         11/25/2019 12:19:05 PM
Machine Name:          comsensus
Machine IP:            10.0.2.15
Machine OS:            Unix 4.9.0.7

Application Domain:    openPDC.exe
Assembly Codebase:     root/openpdc-new/out/openPDC.exe
Assembly Full Name:    openPDC, Version=2.8.40.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.8.40.0
Assembly Build Date:   11/18/2019 1:59:21 PM
.Net Runtime Version:  4.0.30319.42000

Exception Source:
Exception Type:        System.InvalidOperationException
Exception Message:     Failed to initialize web hosting: Could not load type of field 'OSIsoft.AF.PI.PIServer:_channel' (16) due to: Could not load file or assembly 'OSIsoft.PI.Net.Core, Version=1.5.0.0, Culture=neutral, PublicKeyToken=c3309f0734ba2805' or one of its dependencies.
```

ritchiecarroll, 2022-04-01:
Try the latest release:
https://github.com/GridProtectionAlliance/openPDC/releases/tag/v2.9.148

closed 2022-04-01 by ritchiecarroll