GridProtectionAlliance/openPDC #128: OpenPDC Manager 2.7.4.0 - Error during starting up the programm

tmarecek, 2019-12-16:
Hello,

I have instalation of OpenPDC Manager v 2.7.4.0. on Microsoft Windows Server 2016 Standard and following error message is shown up when I want to open OpenPDC Manager. It seems, that output and input streams are working fine. Do you have any suggestion what could cause this behaviour? 

Thank you vey much and have a nice day.

tmarecek, 2019-12-16:
[image]

ritchiecarroll, 2019-12-16:
Commonly this is because you don't have rights to the database openPDC is using. Check the connection string for the openPDC (in the openPDC.exe.config) and see if you have rights to access the specified database.

ritchiecarroll, 2020-01-25:
Going to close this one - let me know if you still have issues.

Thanks!
Ritchie

closed 2020-01-25 by ritchiecarroll