GridProtectionAlliance/openPDC #129: Connection attempt failed for ppa [C:\Program Files\openPDC\Archive\]: Value must be positive

arz-ece, 2020-01-25:
Hello Everyone,

I am new to github, so please bear with me if any mistake I make here!

The issue I have is with archiving OpenPDC streaming data! While I can stream real-time data in OpenPDC, the .d files of PPA and STAT does not get stored anywhere in the relevant OpenPDC folders.

Please find attached a snapshot of the OpenPDC Console that highlights the issue. Also I am copying part of the ErrorLog.txt that gives some information on this:

Application Domain:    openPDC.exe
Assembly Codebase:     C:/Program Files/openPDC/openPDC.exe
Assembly Full Name:    openPDC, Version=2.4.120.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.4.120.0
Assembly Build Date:   3/17/2017 3:04:06 PM
.Net Runtime Version:  4.0.30319.42000

Exception Source:      GSF.Historian
Exception Type:        System.ArgumentException
Exception Message:     Value must be positive
Exception Target Site: set_DataBlockSize

---- Stack Trace ----
   GSF.Historian.Files.ArchiveFileAllocationTable.set_DataBlockSize(value As Int32)
       openPDC.exe: N 00075
   GSF.Historian.Files.FixedTableRegion.ParseBinaryImage(buffer As Byte[], startIndex As Int32, length As Int32)
       openPDC.exe: N 01614
   GSF.Historian.Files.ArchiveFileAllocationTable..ctor(parent As ArchiveFile)
       openPDC.exe: N 00501
   GSF.Historian.Files.ArchiveFile.OpenStream()
       openPDC.exe: N 00914
   GSF.Historian.Files.ArchiveFile.Open()
       openPDC.exe: N 00635
   HistorianAdapters.LocalOutputAdapter.AttemptConnection()
       openPDC.exe: N 00119
   GSF.TimeSeries.Adapters.OutputAdapterBase.AttemptConnectionOperation()
       openPDC.exe: N 00290

(Outer Exception)
Date and Time:         1/25/2020 12:04:13 AM
Machine Name:          RTPIS-DT4
Machine IP:            fe80::29d0:1ee2:6557:f044%13
Machine OS:            Microsoft Windows NT 6.1.7601 Service Pack 1

Application Domain:    openPDC.exe
Assembly Codebase:     C:/Program Files/openPDC/openPDC.exe
Assembly Full Name:    openPDC, Version=2.4.120.0, Culture=neutral, PublicKeyToken=null
Assembly Version:      2.4.120.0
Assembly Build Date:   3/17/2017 3:04:06 PM
.Net Runtime Version:  4.0.30319.42000

Exception Source:      
Exception Type:        System.InvalidOperationException
Exception Message:     Connection attempt failed for ppa [C:\Program Files\openPDC\Archive\]: Value must be positive

Part of the issue comes from the "Value must be positive" message, which I don't understand. 

I truly appreciate if any kind soul could help me with this.

Many Thanks,
Ali

arz-ece, 2020-01-25:
OpenPDCConsole Snapshot attached

StephenCWills, 2020-01-30:
This is usually due to data corruption in the active file. Try deleting the corresponding file from each archive: `ppa_archive.dat` or `stat_archive.dat`.

arz-ece, 2020-01-31:
Thank you for the response, Stephen. I did as you did and while I see the problem with the Statistics Folder solved as it is storing the stat_... .d files, the Archive .d files don't get stored yet! The only files created in the Archive folder are:

scratch.dat
ppa_startup.dat
ppa_dbase.dat2
ppa_archive.d

I am not getting any errors in the Console or  ErrorLog.txt now.

What can be the next step to solve this?

Thanks,
Ali

StephenCWills, 2020-01-31:
Hi Ali,

`ppa_archive.d` is the active file. That's where new data points go as they enter the archive. Once that file gets full, the archive will roll over, essentially renaming that file to include start and end timestamps and then creating a new ppa_archive.d file for the new measurements entering the archive. Therefore, the fact that it is not creating any historic files indicates that the active file is simply not full yet. Whether that is because you haven't waited long enough or there is an underlying error, I couldn't really say. (Note that the active file's size will not increase because the file is preallocated to a specific size when it's created and will likely roll over before it needs to be expanded. Therefore, file size is not a good indicator of whether the archive is working properly.)

If you still feel there might be something worth investigating, there are a number of things you can try. The `list /o` command on the console will show you how many measurements have been archived by the PPA historian, which can give you an idea of whether the historian is actually working. The `list PPA` command will give you a very detailed status of the historian that can give you insights into potential issues. Reinitializing the historian gives you the opportunity to force any errors that may have occurred at startup to reoccur so you can see the error message again.

Feel free to try any of those things I just suggested and report back here if you determine any new information.

Thanks,
Stephen

arz-ece, 2020-01-31:
Dear Stephen,

The Console is running continuously whether OpenPDC is collecting data or not so I am unable to write a command there.

However, PPA looks like below. The interesting thing is the Statistics folder is generating new .d files while the PPA is unable to do that.

stat_archive_2020-01-31 16!42!09.607_to_2020-01-31 18!58!26.615
stat_archive_2020-01-31 14!22!04.104_to_2020-01-31 16!42!09.607

On Console Screen:
[PPA] 42,912,825 measurements have been processed so far...

I have restarted the workstation but no changes...

[image]

[image]

Thanks,
Ali

StephenCWills, 2020-01-31:
You can type commands into the console while it's running. If you're having trouble with that, then you can use the `Remote System Console` page in the openPDC Manager which has a separate text box for entering commands. You can also use the `q` command to help quiet down the console to make it easier to enter commands. When you're done entering commands, use the `r` command to resume.

The stat historian really shouldn't be producing so many new historic files. It sounds like something may be misconfigured in your system. Please double-check that your devices are writing their data to the PPA archive and not the STAT archive.

arz-ece, 2020-01-31:
Stephen, I am reporting back here the new information from executing `list /o` and `list ppa`. Do you detect any problem in it?

[image]

[image]

[image]

[image]

ritchiecarroll, 2020-02-02:
You openPDC version looks rather old - is it possible to upgrade to a newer version? You are on version 2.4, the latest version is 2.7:

> https://github.com/GridProtectionAlliance/openPDC/releases

Before the upgrade I would stop the service and remove the `Archive` and `Statistics` folders.

Also, we generally recommend using the openHistorian for archiving data as this provides many performance benefits, see:

> https://github.com/GridProtectionAlliance/openHistorian

Thanks,
Ritchie

closed 2020-02-05 by arz-ece