GridProtectionAlliance/openPDC #131: Input stream multiple sources question

mlbeauchemin, 2020-04-27:
We had been informed that in version 2.7 of openPDC, there was a way to set up an input stream to have multiple potential sources. I have installed v2.7, but I am unable to figure out how to set this up.

ritchiecarroll, 2020-04-27:
In the connection string for the PMU you can add as many source UP or hostnames as needed, for example:
```
transportProtocol=tcp; server=192.168.2.2:4712,192.168.3.2:4712;
```

ritchiecarroll, 2020-04-27:
Also, just checked - this should be in nightly build version that is pending release for version 2.8 (not 2.7):
http://www.gridprotectionalliance.org/nightlybuilds/openPDC/Beta/Synchrophasor.Installs.zip

Thanks,
Ritchie

closed 2020-04-27 by ritchiecarroll

mlbeauchemin, 2020-06-05:
@ritchiecarroll We've got the version 2.8 code now, would you be able to point us in the direction of where the code behind this is? We are trying to understand it. Thanks.

ritchiecarroll, 2020-06-05:
Code was added to the `GSF.Commuication.TcpClient` to that any connection made with this library could connect to multiple sources.

This works by maintaining an array of servers to connect to and just moving to the next one should it fail to connect with the first:

https://github.com/GridProtectionAlliance/gsf/blob/master/Source/Libraries/GSF.Communication/TcpClient.cs#L642