GridProtectionAlliance/openPDC #135: xbuild issue on Raspberry Pi

steelsurfer, 2020-11-10:
Hardware is RASPBERRY-MODB+-512M. RASPBERRY PI MODEL B+ BOARD
Raspberry Pi OS (32-bit) with desktop and recommended software Version: August 2020

Screengrab shows version of xbuild and mono
buildmono.sh in gsf/Source isn't building as intended
DNP3Adapters.caproj is named Dnp3Adapters.csproj
Is xbuild of filename on Raspberry Pi case sensitive?

./Libraries/Adapters/Dnp3Adapters
./Libraries/Adapters/Dnp3Adapters/Dnp3InputAdapter.cs
./Libraries/Adapters/Dnp3Adapters/Dnp3Adapters.csproj

[image]

StephenCWills, 2020-11-10:
It's actually not xbuild, but rather the file system that is case-sensitive. In this case, the entry in the solution file doesn't match the casing of the file in the Git repo. One of the two would have to be corrected.

steelsurfer, 2020-11-12:
With respect, I've attached install-openPDC.log since I'm reasonably sure that's not the sole issue with building on:-
Raspberry Pi OS (32-bit) with desktop and recommended software Version: August 2020

XBuild Engine Version 14.0
Mono, Version 6.12.0.90

Mono JIT compiler version 6.12.0.90 (tarball Fri Sep  4 14:21:54 UTC 2020)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
TLS:           __thread
SIGSEGV:       normal
Notifications: epoll
Architecture:  armel,vfp+hard
Disabled:      none
Misc:          softdebug 
Interpreter:   yes
LLVM:          yes(610)
Suspend:       preemptive
GC:            sgen (concurrent by default)

[install-openPDC.log](https://github.com/GridProtectionAlliance/openPDC/files/5530016/install-openPDC.log)

ritchiecarroll, 2022-04-01:
Try the latest release:
https://github.com/GridProtectionAlliance/openPDC/releases/tag/v2.9.148

closed 2022-04-01 by ritchiecarroll