GridProtectionAlliance/openPDC #136: OSI PI to Open PDC

Andresd64, 2020-11-11:
Hello, I am trying to connect some measurements in real time from pi system to open pdc, however I have not been able to find more information than that found in the documentation page, I have entered the server values ​​but I do not understand the last setp how to add  measurements manually

ritchiecarroll, 2021-01-26:
Are you trying to read measurements "from" PI into openPDC? Or send data from openPDC "to" PI?

Thanks,
Ritchie

clackner-gpa, 2023-03-09:
Closing this due to inactivity.
For anyone having similar issues we recommend looking at our forum:
https://discussions.gridprotectionalliance.org/search?q=OSISoft%20PI
Feel free to engage over there and we will do our best to support you.