GridProtectionAlliance/openPDC #142: Hardware/OS/DB Benchmarking

jbax3, 2021-11-06:
After a bit of browsing around, I'm having trouble finding the best place to look for IO-rates for different OpenPDF + OS + DB benchmarking.

I've seen the TVA stats of 150 million EPHour, are there any updates to this? Any rough IO cuts to look at for things like a modern blade server, Windows, and MySQL versus same HW, Ubuntu Server, and Postgres?

Related, are there any very active development/usage communities (i.e. Slack) for OpenPDC?

ritchiecarroll, 2021-11-06:
Not sure of any benchmarking studies anyone has done except for us internally, e.g., running 1000 devices with roughly 20 signals at 30 samples per second for stress testing.

But there are some larger production deployments out there running 300,000+ measurements per second - if my math is right, that's about 1,080,000,000 events per hour.

Hope that helps!

Ritchie

jbax3, 2021-11-06:
Thanks, Ritchie! That's very helpful.

Are y'all running one server or multiple to handle 600 EPS? I'm assuming devices are PMUs.

Do you have any thoughts on rough limitations of a Windows+MySQL+SSD+64GB ram vs Ubuntu Server+Postgres+Same HW Stack? In these stacks, do you have any thoughts on where bottlenecks would arise? i.e. MySQL or SSD or ?

Thanks!

ritchiecarroll, 2021-11-06:
Single server - devices are PMUs, yes.

You must be thinking about archiving data - not just measurement routing. Generally speaking we have began deploying openPDC instances for data acquisition and distribution. These days we recommend the openHistorian for archiving data - its engine has much improved over the original openPDC built-in archive feature.

We consider the openPDC time-series archive to be "openHistorian 1.0", and the standalone openHistorian software (see https://github.com/GridProtectionAlliance/openHistorian)  to be "openHistorian 2.0".

Also, the database used with openPDC is only for metadata, i.e., device connections, measurement definitions, etc. You are not going to see much data archive performance with an RDBMS - although people try often. The best performance I've seen has been with a utility that designed time-block BLOB storage with SQL Server - but this same utility helped develop the openHistorian 2.0 ;-)

For Windows vs/ Linux - right now, given the production code is based on the .NET Framework, the Linux builds use Mono - which is traditionally not as fast (but it may be much improved, I've not tried in a year or two). That said, with .NET core, things should be on par, Linux or Windows - and we are already moving in that direction:

https://github.com/gemstone/

Thanks,
Ritchie

jbax3, 2021-11-06:
This makes sense! I'll do some more homework and potentially open up any new questions in openHistorian! Thanks =)

closed 2021-11-06 by jbax3