GridProtectionAlliance/openPDC #145: How can I export the openPDC manager's  data as a CSV file

deymaitreyee, 2022-02-17:
I am able to connect our upmus through openPDC manager. I can see the values in the graph real time measurements. But whenever I tried to see the previous data won't be able to do that. I have seen couple of Q&A but are not related exactly what I want.

[image]

Any help will highly appreciable.
Maitreyee

ritchiecarroll, 2022-02-17:
This depends on if you configured openPDC to archive data when you set it up.

One easy way is to simply install the openHistorian (https://github.com/GridProtectionAlliance/openHistorian) then connect openHistorian to openPDC and use the openHIstorian web interface to export to CSV or COMTRADE.

Thanks,
Ritchie

deymaitreyee, 2022-02-22:
Thanks Ritchie for you reply.

When I configured the openPDC, I have chosen PPA as historian.

I have installed openHistorian manager but couldn't able to link it with openPDC.
Can you help me further on this?

Many thanks
Maitreyee

deymaitreyee, 2022-02-22:
Hi 

Additionally, now i have managed to synch the OpenHistorian with openPDC but can't getting any graph measurements. see below pic.
[image]

How can i see the same graph measurements?
How can I get the data as a csv file?

Many thanks
Maitreyee

ritchiecarroll, 2022-02-22:
Can you enumerate the steps you followed to connect openHistorian to openPDC?

I would expect something like following:

_Note: You may need to delete existing `PDC` connection before following these steps._

1) From within the openHistorian Manager, go to "Inputs > Subscription Based Inputs > Create Internal Subscription"
2) Enter `PDC` in the "Acronym" field
3) Enter `localhost` in the "Hostname" field
4) Enter `7165` in the "Port" field
5) Uncheck "Use Source Prefix"
6) Click "Next" button
7) Click "Save" button

One this is working properly, you can browse to http://localhost:8180/ to trend and export data to a CSV.

Thanks,
Ritchie

closed 2022-03-22 by ritchiecarroll