GridProtectionAlliance/openPDC #146: Connecting opendc manager

bmitrauncc, 2022-03-18:
Hi,

I am having trouble connecting my device via TCP to openpdc manager, everything works properly when I use the PMU connection tester. 

When I request configuration I receive a "Timeout message", but I have been saving the config and connection file from PMUConnection tester. I am using them to load my device configuration.

[image]

Along with that I wanted to know once I stream data in the OpenPDC manager how do I save it?

ritchiecarroll, 2022-03-19:
Looks like something is not right :-) If you can connect with PMU Connection Tester, you should able to connect with openPDC - they use the same code base.

A timeout sounds odd - maybe your openPDC is not setup correctly or the service is not running?

Can you open Windows Services, find `openPDC` in the list and see if the service is running?

Per your question on how to save data, one easy/good way is [openHistorian](https://github.com/GridProtectionAlliance/openHistorian)

You can connect openHistorian to openPDC so that it will archive all its data.

Thanks!
Ritchie

bmitrauncc, 2022-03-21:
Hi RItchie,

I tried but error seems to be same, I am attaching some screenshots for your reference:

[image]
[image]
[image]

Kindly requesting your assistance.

ritchiecarroll, 2022-03-21:
Try switching protocol to 2005 instead of 2011

bmitrauncc, 2022-03-21:
Just tried, got this error: But the connection with PMU connection tester was successful

[image]

ritchiecarroll, 2022-03-22:
Two things:

1) I notice in the connection string you are specifying a specific IP for traffic, does the connection work better if you specify the default interface, i.e., `; interface=0.0.0.0`?
2) Also, is the PMU Connection Tester connected and running while you are trying to use the device wizard in openPDC? If so, stop the PMU Connection Tester connection first, then attempt to request the configuration - some devices only support a single connection.

Thanks,
Ritchie

bmitrauncc, 2022-03-22:
Hi Ritchie,

I just tested and your second option worked for me... Thank you so much for the help....

I have an additional follow-up question, once I run the openHistorian it automatically archives data correct? And how do I extract it for post processing?

Thanks,

Bhaskar

ritchiecarroll, 2022-03-22:
If you have connected the openHistorian connected to the openPDC (see instructions below) or you have directly connected openHistorian to your device (note that it looks like only one system can connect to device at once), then yes, openHistorian will be archiving data automatically.

Browse to http://localhost:8180 and login to openHistorian, then click on Trend/Export data. Select the desired measurements and time window, then can trend (visualize to UI) or export to CSV or COMTRADE.

Hope that helps!

Thanks,
Ritchie

_Instructions on connecting openHistorian to openPDC_: FYI, the following link is for SIEGate, but the exact same steps apply to openHistorian:

https://github.com/GridProtectionAlliance/SIEGate/blob/master/Source/Documentation/wiki/Creating_Internal_Gateway_Connections.md

ritchiecarroll, 2022-03-22:
BTW - I am going to close this issue. If you have other general questions such as these, I suggest you visit our discussions board:

https://discussions.gridprotectionalliance.org/c/gpa-products/openpdc/6

Thanks,
Ritchie

closed 2022-03-22 by ritchiecarroll