GridProtectionAlliance/openPDC #159: OpenPDC install problem

snail526, 2023-03-30:
hello everyone,

I ran into the following issue while installing openPDC and would like to be resolved. There is no test device information and data in the openPDC Manager (indicated in red in the figure). I want to determine if the openPDC installation is correct based on the test setup information. 

Hope to hear from you, thank you.
[image]

snail526, 2023-03-31:
Hi community,

Thank you for your reply, I have successfully connected the test device. I still have some questions about how to use the adapter to transfer data from openPDC to kafka, and  hope to receive your professional answers. Mostly about adapter parameter configuration, as shown in the figure below, I want to know how these parameters should be configured,so that I can see these data in kafka.

Best wishes,
ming

从 Windows 版邮件发送

发件人: Stephen C. Wills
发送时间: 2023年3月30日 20:57
收件人: GridProtectionAlliance/openPDC
抄送: snail526; Author
主题: Re: [GridProtectionAlliance/openPDC] OpenPDC install problem (Issue#159)

Hello snail256,
Did you set up the openPDC using the Sample Data Set? That would have provided configuration for the SHELBY test device automatically. I assume you didn't, so if you'd like to add it via openPDC Manager, try using the Input Device Wizard.
There should be a Sample1344.PmuCapture file in the openPDC folder provided with the installation. So you'd select the File protocol in the first step, browse to that file, and select IEEE 1344 as the protocol. In the second step, use the Request Configuration button to retrieve the device configuration. In the last step, make sure to enter all phase values as + for positive sequence.
When you finish the wizard, the openPDC should now have an input device defined for which it will start processing test data from the sample file.
Thanks,
Stephen

snail526, 2023-04-03:
> 

Hi community,

Thank you for your reply, I have successfully connected the test device. I still have some questions about how to use the adapter to transfer data from openPDC to kafka, and  hope to receive your professional answers. Mostly about adapter parameter configuration, as shown in the figure below, I want to know how these parameters should be configured,so that I can see these data in kafka.

Best wishes,
ming

snail526, 2023-04-03:
Hi community,

Thank you for your reply, I have successfully connected the test device. I still have some questions about how to use the adapter to transfer data from openPDC to kafka, and  hope to receive your professional answers. Mostly about adapter parameter configuration. I want to know how these parameters should be configured,so that I can see these data in kafka.

Best wishes,
ming

StephenCWills, 2023-04-03:
Hi ming,

Here is the latest info on the Kafka adapter. Which is to say that it likely will not work with your installation of Kafka.
https://discussions.gridprotectionalliance.org/t/kafka-producer-configurations/969

TBH, I don't have any working examples of configuration as we've only ever set up the adapter on-site for a single user. Your best bet will be to review the source code in GSF.
https://github.com/GridProtectionAlliance/gsf/tree/master/Source/Libraries/Adapters/Kafka

In particular, have a look at properties with the `ConnectionStringParameter` attribute in `TimeSeriesProducer` to see what parameters you can provide when setting up the adapter. You should also be able to see much of the same information on the `Custom Output Adapters` page of openPDC Manager.

Thanks,
Stephen