GridProtectionAlliance/openPDC #168: High memory usage cause by Statistics Archive

jjneuha, 2024-02-23:
One of our OpenPDC instances (2.9.42.0) experienced a strange issue last night into this morning. Starting around 22:00 last night, all of our subscribers quit receiving data.  When I pulled up openPDC Manager this morning, it showed memory usage was at 99% and times were off by 51s. We had patched Windows yesterday with the latest cumulative updates, so my hunch was it had something to do with the patches. I uninstalled the patches and rebooted several times, yet I still observed the openPDC memory steadily climbing in Task Manager until it consumed everything. I started looking through the log files and noticed the following:

[image]

I disabled the Statistics Archive and restarted the service and that seemed to resolve the memory issue for now. 

[image]

My question is, how should I go about identifying what the root issue is? Nothing in our configuration has changed for over a year, so I'm at a loss to explain what would have initiated this problem.

ritchiecarroll, 2024-02-23:
Looks like the `stat_archive.d` got corrupted during the restart. This is a known, rare failure mode we've seen this before with the 1.0 historian. You can correct by stopping `openPDC`, deleting the `stat_archive.d` file and restarting the `openPDC` service. You can then safely re-enable the statistics.

If you were able to possibly share the `stat_archive.d` (before deleting) this could assist us in replicating / correcting the issue, as this is difficult to replicate in a development environment. If this is a possibility, you could upload here:

https://upload.gridprotectionalliance.org/upload/CorruptedStatArchive

Thanks,
Ritchie

jjneuha, 2024-02-23:
@ritchiecarroll , unfortunately I had already decided to try deleting those files before I read your reply. That did fix the issue and I was able to re-enable stats, but now I do not have a sample of stat_archive.d to upload to you. I'll close this issue since it is already a known problem. If it happens again, I will send you a copy of the archive.

closed 2024-02-23 by jjneuha